mantidproject/mantid #129: Invalid NeXus file after "Save Nexus"

rayosborn, 2015-01-19:
After saving a Mantid Processed Workspace to a NeXus file, using the "Save Nexus" algorithm, the resulting file had the following data structure (ignoring all the other groups):

    test:NXroot
      @HDF5_Version = 1.8.9
      @NeXus_version = 4.3.0
      @file_name = /Users/rosborn/test.nxs
      @file_time = 2015-01-19T10:27:34-06:00
      mantid_workspace_1:NXentry
        definition = Mantid Processed Workspace
          @URL = http://www.nexusformat.org/instruments/xml/NXprocessed.xml
          @Version = 1.0
        definition_local = Mantid Processed Workspace
          @URL = http://www.isis.rl.ac.uk/xml/IXmantid.xml

...

    title = LaPd2Al3 24.53g 60meV 200Hz T=6K
    workspace:NXdata
      axis1 = float64(286)
        @distribution = 1
        @units = MomentumTransfer
      axis2 = [-50.  -49.5 -49.  ...,  53.5  54.   54.5]
        @units = DeltaE
      errors = float64(210x285)
      frac_area = float64(210x285)
      values = float64(210x285)
        @axes = axis2,axis1
        @signal = 1
        @unit_label = Counts per microAmp.hour
        @units = Counts

The main issue is that the units attributes of 'axis1' and 'axis2' are being used to store the type of the axis, i.e., MomentumTransfer and DeltaE, not the physical units, e.g. Angstrom-1 or meV. 

A related issue is that it would be helpful to the end user if the axis names were more descriptive:

    title = LaPd2Al3 24.53g 60meV 200Hz T=6K
    workspace:NXdata
      momentum_transfer = float64(286)
        @distribution = 1
        @units = Angstrom-1
      energy_transfer = [-50.  -49.5 -49.  ...,  53.5  54.   54.5]
        @units = meV
      errors = float64(210x285)
      frac_area = float64(210x285)
      data = float64(210x285)
        @axes = energy_transfer,momentum_transfer
        @signal = 1
        @unit_label = Counts per microAmp.hour
        @units = Counts

There are other Mantid Workspaces that have worse problems, with axes being defined by XML strings, but I'll try to document that in another issue.

NickDraper, 2016-02-15:
Ross what more is there to do for this?

rayosborn, 2021-02-24:
Was any progress made on this issue or has it been superseded by other developments?

closed 2021-08-30 by stale[bot]

rayosborn, 2021-08-30:
Mantid continues to produce NeXus files that do not correspond to the NeXus standard. Indeed, recent MDHistoWorkspace files have a multitude of issues, such as invalid field names (e.g., '[H,0,0]'), persistent use of long-deprecated signal and axis attributes (group attributes have been recommended for ten years now), improper definition of error fields, and failure to put weights in even the same file as the signal. Reading such files in NeXpy is a continual headache, even though NeXpy is designed to be tolerant of non-conforming files. This makes MDHistoWorkspace files almost useless outside Mantid for all but the most persistent experts. A proper NeXus-conforming export function is long overdue.