mantidproject/mantid #157: RPM dependencies not resolvable using copr

zjttoefs, 2015-02-02:
On a fresh fedora 20 with the copr repo enabled according to instructions nothing gets installed. It just ends with this error:

==> default: --> Finished Dependency Resolution
==> default: Error: Package: mantid-developer-1.8-2.fc20.noarch (mantid-mantid)
==> default: Requires: rpmfusion-nonfree-release
==> default: Error: Package: mantid-developer-1.8-2.fc20.noarch (mantid-mantid)
==> default: Requires: python-ipython >= 1.1
==> default: Available: python-ipython-0.13.2-2.fc20.noarch (fedora)
==> default: python-ipython = 0.13.2-2.fc20
==> default: Available: python-ipython-0.13.2-4.fc20.noarch (updates)
==> default: python-ipython = 0.13.2-4.fc20

To the uneducated eye that looks as if rpmfusion-nonfree-release is missing and python-ipython is only available as 0.13.2-X instead of the expected >=1.1.

Creating ticket here, due to lack of trac account.

sic, 2015-02-02:
I'm adding the rpmfusion free/nonfree release rpms into the mantid copr.  I've also added the ipython rpms.

I'll let you know when everything is built.

zjttoefs, 2015-02-03:
Now it manages to install stuff. Nice. 

Some things are missing according to cmake:
* doxygen (non fatal, but I would include that for developers)
* squish (is that used? If so developer should have it.)
* sphinx_bootstrap_theme (fatal)

sic, 2015-02-03:
The sphinx_bootstrap_theme is giving me problems on copr, I have a standalone rpm that works.  Will let you know when I beat copr into submission.

peterfpeterson, 2015-02-03:
@zjttoefs squish is optional (and commercial software). `doxygen` is a good dependency to add, but I feel like we should wait until we have `sphinx_bootstrap_theme` squared away before the `mantid-developer` rpm gets updated.

stuartcampbell, 2016-01-15:
I think that this is all sorted now, if I'm mistaken please reopen.

closed 2016-01-15 by stuartcampbell