mantidproject/mantid #12742: Forward declare headers for some workspace types

peterfpeterson, 2015-06-04:
Create forward declares of common workspace types. Explicitly:
* IEventWorkspace
* IMDEventWorkspace
* IMDHistoWorkspace
* IPeaksWorkspace
* ITableWorkspace
* MatrixWorkspace
* Workspace

The headers will all be labeled something like `ITableWorkspace_fwd.h` and have the form of
```c++
namespace Mantid {
  namespace API {
    class ITableWorkspace;
    typedef boost::shared_ptr<ITableWorkspace> ITableWorkspace_sptr;
    typedef boost::shared_ptr<const ITableWorkspace> ITableWorkspace_const_sptr;
  }
}
```

eXeC64, 2015-06-04:
Since there's so few of them and no dependencies, would it make sense to put them all in a single `Workspace_fwd.h` header? That way when writing a header, you can just include `Workspace_fwd.h` and not have to think too hard, and in the cpp file, include the required real headers.

peterfpeterson, 2015-06-04:
That feels like a good modification. Also it will have a lot less repeated junk in the `_fwd.h` files.

PR #12743 (2015-06-05): Added forward only headers for common workspaces.
Fixes #12742.

**To test:** If the build servers pass it is good.
closed 2015-06-05 by quantumsteve