mantidproject/mantid #12827: Extend Shape classes to support triangle based shapes

NickDraper, 2015-06-16:
This is to support the ability to load in scanned images of sample geometries.

This will require inheriting from object or surface class and providing a class that can support all the same methods but using a triangular based mesh as the source data.

NickDraper, 2017-07-14:
So After a quick look I suggest the following steps:

1) Create an IObject.h class as a base interface to the Object Class (empty for now)
2) Rename Object to CSGObject, and make it inherit from IObject
3) Change ObjComponent to only use IObject, and return IObject from shape()
4) See what fails this will give you the minimal methods you need to define on IObject.h
5) Create a class MeshObject that implements IObject
6) Write plenty of tests

NickDraper, 2017-07-14:
The methods that appear to be needed on IObject include:

* isValid
* isOnSide
* interceptSurface
* solidAngle
* getBoundingBox
* getPointInObject
* draw
* initDraw
* HasValidShape
* Whatever Link::object needs

KarlPalmen, 2017-11-07:
What does CSG in CSGObject stand for?

NickDraper, 2017-11-07:
Constructive Solid Geometry https://en.wikipedia.org/wiki/Constructive_solid_geometry

NickDraper, 2017-11-07:
So this is something you need to decide yourself, but you will need to:

1. Load STL files #12828 
2. Provide a geometry mesh to the rendering code of the instrument display (look at the *Renderer.cp files in Geometry\rendering).
3. implement the methods in the comments above

You might want to consider if any of the libraries we already use (OpenCascade, OpenGL) have anything to help out here, or look at other libraries (e.g. Qhull) that might help.

KarlPalmen, 2017-11-08:
I found out that STL lists each of  its triangles simply as the coordinates of its vertices rather than referring to a list of vertices. All three of the polygon mesh formats mentioned in Wikipedia article on polygon meshes, have a list of vertices. This suggests either Face-Vertex mesh or Winged-Edge mesh, because they both have a face list. The geometric relationships between the triangles would need to be inferred by the STL reader by comparing vertices for equality.

KarlPalmen, 2017-11-08:
I've attempted the first part of step 2, but it seems that Visual Studio on my machine is not up to the task of renaming it everywhere it does need renaming. The very large number of references to Geometry::Object makes this a non-trivial task, for which I'll need help.

KarlPalmen, 2017-11-09:
A second attempt was successful in renaming the class. The files still need to be renamed.

KarlPalmen, 2017-11-14:
In suggestion 3 (of July 14 comment), the 'only' is not precise enough for me into understand. I ask "Instead of what?". It seems to me to be both ``IObjComponent`` , ``CSGObject`` and nothing else.

KarlPalmen, 2017-11-14:
I'm now working on replacing ``CSGObect`` with ``IObject``.

KarlPalmen, 2017-11-14:
After steps 3. & 4. have been completed, I'll put the branch out in a pull request and when this is approved and merged, I'll have a good foundation for the MeshObject of step 5.

KarlPalmen, 2017-11-16:
In doing step 4, I've found a lot fails in many parts of Mantid. I've got the Geometry project to build OK and GeometryTest builds OK except for 2 link errors and there are errors in other parts of Mantid.

I've become aware that the underlying problem is what should be ``Iobject`` and what should be ``CSGObject``. I think as much as possible should be ``Iobject`` so we don't tell the code how we implement the object, but when it comes to constructing an object, the constructor will at some point need to know whether to construct a ``CSGObject``, ``MeshObject`` or some other class inheriting from Iobject.

KarlPalmen, 2017-12-08:
Reopened to take MeshObject class.

KarlPalmen, 2017-12-15:
I chose the following shapes for testing: cube (simple), octahedron (faces & edges not parallel to axes, yet easy to work out whether a point is inside/outside it etc.) & L-shape (not convex).

NickDraper, 2018-01-04:
If you decide on the code it yourself option these are worth looking at:

**isValid**
```
Pseudo code
Ray R = Ray(P, Vector(1,0,0));
int intersections = 0;
foreach(Triangle T in your Mesh) {
    if(R intersects T)
        intersections++;
}
if(intersections % 2 == 1) {
    // INSIDE
} else {
    // OUTSIDE
}
```
The intersections code can use the interceptSurface code

**isOnSide**
As for "exactly on the surface", you should code a special case in the ray-triangle intersection. If the intersection is only a short way from the beginning of the ray, you can count this as "on the surface", and abort the whole algorithm.

**interceptSurface**
Google for `ray-triangle intersection`
I found this one, but there may be better
https://stackoverflow.com/questions/28165548/ray-triangle-intersection-c
You will need to loop over all the triangles

**solidAngle**
The existing code for CSGObject uses a triangulated surface you should be able to use that

**getBoundingBox**
Loop over all of the vertices, get the largest, make a box

**getPointInObject**
Options:

1. pick the centre point of the bounding box
1. pick a vertex and go slightly inwards
1. Test whatever you have picked with isValid, if fail go for the other option

**draw**
Should be easy for triangular meshes

**draw**
Should be easy for triangular meshes

**HasValidShape**
Not sure what this one does

KarlPalmen, 2018-01-05:
Thank you @NickDraper for this. Over the course of looking for code to help with this. I did also find at least one website helpful for writing my own code.

KarlPalmen, 2018-07-19:
Closed, because changing base branch stopped it from being closaed automatically.

PR #21245 (2017-12-07): Make space for MeshObject
The geometry objects are rearranged to enable a MeshObject to be defined.

IObject class is defined. Object is renamed CGSObject and made to inherit from IObject.
ObjComponent is modified to use IObject instead of CSGObject.

**To test:**

Code Review and you may check that object can be rendered in instrument view or elsewhere.

Fixes #12827, except that MeshObject still needs to be added. The merger of this change beforehand will help with this.

CSGObject has a performance test. Run it (as the Object performance test) without this fix and with this fix to see wheth [... cut]
closed 2018-07-19 by KarlPalmen