mantidproject/mantid #12828: LoadSampleGeometry algorithm

NickDraper, 2015-06-16:
create an algorithm that will load a STL sample geomtery file and assign that geometry to the sample.

sample files can be found here
http://files.mantidproject.org/videos/StlFiles/

KarlPalmen, 2018-02-22:
The picture appears to be a customized dialog box, which would not arise by default when writing an algorithm. I'll need help about this.

KarlPalmen, 2018-02-22:
Also, I don't understand the context of the algorithm. Shouldn't a workspace be involved?

NickDraper, 2018-02-22:
Yes, good point about the workspace, yes it needs a workspace to get the instrument.

NickDraper, 2018-02-22:
For the enabling / disabling based on another property you can use setPropertySettings.  Take a look at line 107 of ConvertToMD.cp for an example.

PR #22016 (2018-05-02): LoadSampleShape algorithm for STL files
Description of work.

Creation of new algorithm to load and STL file into a MeshObject and place the resulting shape into the sample of the output workspace.

**To test:**

* Careful code review (issue #12828 only).
* Check that unit tests work and are adequate.
* Run the algorithm in Mantid to see that the dialog box is OK and it doesn't crash.
* Build and check the documentation.

Fixes #12828. 

**Release Notes** 
In New Algorithms of Framework in the release notes of 3.13.

---

#### Reviewer ####

Please comment on the following ([full description]( [... cut]
closed 2018-05-02 by martyngigg