mantidproject/mantid #12832: improve auto-select normalization for MDHistoWorkspace

quantumsteve, 2015-06-16:
Setting NumEventsNormalization for diffraction returns an incorrect result. @AndreiSavici suggests that for MDHistoWorkspace, NumEventsNormalization should only be used if the instrument is inelastic and there is no MDNorm algorithm in the history. 

I think the best place to implement this logic is in [MDHistoWorkspace::displayNormalization()](https://github.com/mantidproject/mantid/blob/master/Code/Mantid/Framework/DataObjects/src/MDHistoWorkspace.cpp#L1231).

OwenArnold, 2015-06-17:
@Andrei, @quantumsteve. I have some questions. Is this blocking the move to newer ParaView in any way? Secondly, yes as steve suggests, we can customise MDHistoWorkspace::displayNormalization(). Rather than trying to parse history looking for MDNorm, should we set this property appropriately as part of MDNorm ?

quantumsteve, 2015-06-17:
@OwenArnold This is NOT blocking our move to newer ParaView. At the same time, this needs to be addressed before it affects a substantial number of users.

AndreiSavici, 2015-06-17:
@quantumsteve , @OwenArnold  I like the idea of setting the flag as part of MDNorm. Then the default should be none. ConvertToMD should check in the inelastic case if you start from a workspace2D (NumEventsNormalization) or events (NoNormalization)

OwenArnold, 2015-06-17:
@AndreiSavici. Ok sounds like a plan. When I get this done I'll assign you as the tester.

OwenArnold, 2015-06-17:
@quantumsteve I'll up the priority of this. I'm working on another critical issue at the moment, but I'll put this next in line.

OwenArnold, 2015-07-06:
@AndreiSavici. Could you please clarify the rules for this?

1)
> NumEventsNormalization should only be used if the instrument is inelastic
Is this really technique dependent?

2)
> and there is no MDNorm algorithm in the history
Are we just trying to determine if Normalization has already been applied?

AndreiSavici, 2015-07-06:
@OwenArnold NumEventsNormalization was the only way to get S(Q,w) for inelastic scattering, and it relies on the matrix workspace going into ConvertToMD to be a Workspace2D histogrammed in energy transfer. The number of events in this case is not neutron events, but number of histogram bins corresponding to scattering in a particular Q,w MDBox. If you do diffraction, you measure only S(Q), so no energy transfer. If you use events in the ConvertToMD, the number of events normalization yields always 1 (unless you weighted the events).
On a regular grid, the MDNorm algorithms correctly calculate the normalization, so you can measure your S(Q,w) with events (not carry around a lot of empty bins). Even if one measures with histograms in w, the MDNorm yields a similar number to number of events normalization, so you should not normalize twice.

OwenArnold, 2015-07-06:
In ConvertToMD:
1. Check that we have an Input Workspace2D
 1. Check that we are converting in InElastic mode 
  1. Set the Normalization to NumEvents
1. Otherwise set the normalization to be None

In the MDNorm Normalization algorithms. Overwrite Normalization type to be None. Already accounts for number of events.

AndreiSavici, 2015-07-06:
In ConvertToMD check first if the mode is elastic. If yes, set to volume normalization (so you can see Bragg peaks in diffraction). If no, check if input is Workspace2D. If yes, set to # events normalization, else set to no normalization. MDNorm is OK

OwenArnold, 2015-07-09:
@andrei. Regarding what you've said above. For diffraction volume normalization is a good idea when you have an MDEventWorkspace (box size varies). When you bin your diffraction workspace to a regular grid MDHistoWorkspace, then volume normalization has no effect other than to uniformly scale the data.

Also, for MDEventWorkspaces shouldn't we always be normalizing by volume?

Sorry to be a pain about this, but I want to make sure we agree on the rules for the default.

OwenArnold, 2015-09-11:
Steps to complete:

1. ~~Make the MDHistoWorkspace take and return the display normalization as a setting~~
1. ~~Make MDHistoWorkspace persist the above information~~
1. ~~Make the MDEventWorkspace able to carry the display normalisation it will pass on~~
1. Make the MDEventWorkspace persist the information to pass to the MDHistoWorkspace
1. Use the rules outlined by @Andrei above to modify ConvertToMD such that generated MDEventWorskpaces are set with the display normalization settings they will pass on to future MDHistoWorkspaces
1. In MDNormDirectSC, set the normalization on the output MDHistoWorkspace to be None.

**The next task would then be to make the SliceViewer query the displayNormalization of input workspaces and use it to set the default normalization** (probably a separate ticket).

PR #13763 (2015-09-29): Visual normalization enhancement for MD Workspaces
Fixes #12832 

__Context__ 

With this ticket we want to ensure that MDHisto workspaces show the correct normalization when displayed in the VSI

The rules for MDHIsto normalization is:
IF  Is not based on Q3D or QMod:
  VOLUME_NORMALIZATION
ELIF energy mode is Elastic:
  VOLUME_NORMALIZATION
ELIF  underlying workspace was event worksapce:
   NO_NORMALIZATION
ELSE 
   NUMEVENT_NORMALIZATION

## For Testing

__Check ConvertToMD follows the rules specified above__
```python
ws2D = CreateSimulationWorkspace(Instrument='MAR',BinParams=[-3,0.1,3],U [... cut]
closed 2015-09-29 by AndreiSavici