mantidproject/mantid #12856: Splatterplot should the display sources when the underlying workspace is changed

picatess, 2015-06-18:
@VickieLynch , @OwenArnold 

__Context__

A source in the VSI should be updated when the underlying workspace changes. This apparently works for the Standard View mode but not for the Splatterplot View mode.

This issue was reported by @VickieLynch  who was using SumMD on a workspace which had a source representation in the Splatterplot view mode. Yet the source did not update when she applied the algorithm.  

__Directions to investigate__

The responsible part of the code is [here](https://github.com/mantidproject/mantid/blob/master/Code/Mantid/Vates/VatesSimpleGui/ViewWidgets/src/MdViewerWidget.cpp#L1376:L1400). This should in principle also work for the Splatterplot mode. 

Things to check:
*  Is the workspace picked by: https://github.com/mantidproject/mantid/blob/master/Code/Mantid/Vates/VatesSimpleGui/ViewWidgets/src/ViewBase.cpp#L731:L756 ?
*  Does the underlying plugin source in the Splatterplot (the one which is invisible) reload the data?
*  Is that change propagated up the pipeline to the Splatterplot filter?

picatess, 2015-06-18:
Regarding 1. The workspace is being detected and the RequestData method is called on the underlying source.

picatess, 2015-06-19:
Thinking about it: I am not sure that the image of the Splatterplot has to change when the underlying source changes. In the Splatterplot we specify the number of points and the percentile to show, so added points do necessarily have to show up.

OwenArnold, 2015-06-19:
@picatess Yes. If you want to verify this you could make the threshold more inclusive and watch the number of points change on the information tab.

AntonPiccardoSelg, 2017-10-05:
@quantumsteve I am reassigning this to you. Please feel free to change this, if this is not appropriate.

closed 2021-03-02 by stale[bot]