mantidproject/mantid #12871: IPython notebook broken on OSX 10.9

OwenArnold, 2015-06-19:
Here's the user report:

```
ImportError: No module named jsonschema

    IPython notebook format depends on the jsonschema package:
    
        https://pypi.python.org/pypi/jsonschema
    
    Please install it first.

I can bypass the error by installing the missing package by sudo pip install jsonschema

However when starting the notebook server I get the following:

dyld: Symbol not found: __cg_jpeg_resync_to_restart
  Referenced from: /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libTIFF.dylib
  Expected in: /Applications/MantidPlot.app/Contents/MacOs//libJPEG.dylib
 in /System/Library/Frameworks/ImageIO.framework/Versions/A/Resources/libTIFF.dylib

This causes a failure to import mantid.simpleapi.   See attached screen grab. 
```

quantumsteve, 2015-06-19:
Updated list of dependencies
https://ipython.org/ipython-doc/3/install/install.html#dependencies-for-the-ipython-html-notebook

quantumsteve, 2015-06-19:
@OwenArnold The second portion of the report suggests that this user still has `DYLD_LIBRARY_PATH= /Applications/MantidPlot.app/Contents/MacOS` set in his or her `.bash_profile`. This causes an issue because the system library libTIFF.dylib is picking up the wrong libJPEG.dylib. (the one included with Mantid is a HDF4 dependency built by homebrew, NOT the system library).

PR #12886 (2015-06-22): Add missing dependencies for ipython notebook.
This fixes #12871.

IPython notebook was failing because of missing dependencies. After upgrading IPython notebook, it looks like new dependencies have been added. According to pip, the full list is 

* gnureadline
* tornado>=3.1
* pyzmq>=2.1.11
* jinja2 
* certifi
* backports.ssl-match-hostname
* markupsafe

The file `make_package.rb.in` has been updated to copy over those dependencies. 

testing:
1. Download Mantid app bundle from the build server or build your own from source.
2. Run IPython notebook according to the [instructions](http://www.m [... cut]
closed 2015-06-22 by OwenArnold