mantidproject/mantid #12882: Add methods for peak memory usage Kernel::Memory

peterfpeterson, 2015-06-19:
Incorporate code from [getRss.c](http://nadeausoftware.com/articles/2012/07/c_c_tip_how_get_process_resident_set_size_physical_memory_use) into [Kernel::Memory](/mantidproject/mantid/blob/master/Code/Mantid/Framework/Kernel/inc/MantidKernel/Memory.h) so peak memory usage could be added to tests.

OwenArnold, 2015-06-22:
@peterfpeterson What exactly is the use case for this? What sort of tests do you want to add this capability to?

peterfpeterson, 2015-06-22:
@OwenArnold It would have helped me know about #12880 closer to when it got introduced. Now it will help me track down when reductions take a long time, if it is because peak memory is large.

OwenArnold, 2015-06-22:
@peterfpeterson I'm just wondering how/if we can use this sensibly in an automated way. I think the feature is definitely useful, I'm trying to establish how we best use it.

PR #12903 (2015-06-23): Kernel::Memory addition
Fixes #12882. This does not require any changes to the release notes.

**To test:** This will either compile everywhere or it won't.
closed 2015-06-23 by eXeC64