mantidproject/mantid #12916: Fix axes of scaled data sets in the VSI

picatess, 2015-06-24:
When scaling a source in the VSI in the Standard View mode, we do not want the axes to be scaled. Currently they are being scaled as well. This appears to be a migration issue. The relevant classes in PV which set the axes are (vktCubeAxesRepresentation)[http://www.paraview.org/ParaQ/Doc/Nightly/html/classvtkCubeAxesRepresentation.html] and (vtkPVChangeOFBasisHelper)[http://www.paraview.org/ParaView/Doc/Nightly/www/cxx-doc/classvtkPVChangeOfBasisHelper.html]

closed 2015-06-24 by OwenArnold

quantumsteve, 2015-06-24:
@picatess Will this work with the new [Axes Grid notation](http://www.kitware.com/blog/home/post/871)?

picatess, 2015-06-24:
@quantumsteve I think it might not work with the Axes Grid notation, as we need to make use of vtkCubeAxesRepresentation. Not sure if this is still used or not.

But the old implementation was outdated and the keywords were not used any longer within ParaView. As pV makes use of vtkArrays which are added as FieldData, ParaView didn't complain about the stuff we were adding ( you can add whatever you like in fact). 

Now we have an implementation which does not set the FieldData directly, but rather uses PV methods to do so. So should this way of implementing stop working, then I expect Mantid to crash and let us know that we need to change something, rather than staying stumm about it.

PR #12918 (2015-06-24): Fix axes of scaled data sets
Fixes #12916 

__For testing__

1. Load a sample data set into the VSI 
2. Go to the Standard View mode
3. Enable the "Show Axis" checkbox and make a note of the extents
4. Press the scale filter; set the x-scale to 2; press apply
5. Enable the "Show Axis" checkbox
  * Confirm that the extent in the x-direction has not changed
6. Change the scales in the other on-the-fly
  * Confirm that the extents do not change