mantidproject/mantid #12927: Add a "peaks diagnostics" interface in the EnginX GUI

FedeMPouzols, 2015-06-24:
When we have a first version of the EnginX GUI #12610, add the peaks visualization and diagnostics dialog.

This is blocked/conditional on:
- [ ] #12610

This shows the peaks (individually) on a graph. A possible simple layout is: list of peaks in a table view on the left + plot area on the right. In principle this is a modal dialog that is shown when the user does a calibration, and when you accept it, the results from that calibration are saved.

FedeMPouzols, 2016-03-04:
@shahrozahmed1 : this issue had been created several months ago based on early discussions with Engin-X scientists. The more up to date requirements are described in this mock-up that you prepared recently: #15374.

DavidFair, 2016-08-04:
Design Documentation link: `https://github.com/mantidproject/documents/blob/master/Design/ENGIN-X/GUI/Engineering_Diffraction_Diagnosis_Dialog_Mockup.zip`

closed 2021-03-03 by stale[bot]