mantidproject/mantid #12967: FindPeaksMD Performance Problems

OwenArnold, 2015-06-29:
There has been a huge drop in the performance since v3.4 of Mantid. I just ran [this]( https://github.com/mantidproject/mantid/blob/master/Code/Mantid/scripts/Vates/SXD_NaCl.py) and the FindPeaksMD step took **30 seconds** I downloaded version 3.4 and the same algorithm took only **0.5 seconds** to run.

1. Fix the x60 decrease in performance. 
1. Add Performance tests to this algorithm so that we can spot any future regressions.

My testing was done on OSX 10.10

peterfpeterson, 2015-06-29:
This is a duplicate of #12713.

peterfpeterson, 2015-06-29:
[This logging statement](https://github.com/mantidproject/mantid/blob/master/Code/Mantid/Framework/MDAlgorithms/src/FindPeaksMD.cpp#L165) calls a bit of code in `Instrument` that explicitly mentions it is slow.

OwenArnold, 2015-06-29:
@peterfpeterson: For the record, we discussed this, and don't think this is a duplicate of #12713 as the comment above suggests.

OwenArnold, 2015-07-14:
Already merged

closed 2015-07-14 by OwenArnold