mantidproject/mantid #12994: MDEventWorkspace incomplete copy/clone.

SimonHeybrock, 2015-07-02:
The copy constructor of `MDEventWorkspace` (and thus `MDEventWorkspace::clone()`) does not seem to cover the case of a file-backed workspace.  If cloning is done with the `CloneMDWorkspace` (or `CloneWorkspace`) algorithm there is a special treatment for this case.

This special treatment should probably be moved to the copy constructor.

As a consequence the will allow us to cleanup `CloneWorkspace` and `CloneMDWorkspace`, since there are no more special cases based on the result of a `dynamic_cast`.

See also #12951.

NickDraper, 2016-02-15:
Pete, you may want to assign on to your current eventworkspace expert.

BhuvanBezawada, 2018-06-07:
@SimonHeybrock 
Hi Simon - I was just having a go at trying to use the code from the CloneMDWorkspace file in the Copy Constructor of MDEventWorkspace as you mention above. The problem is that I cannot access the `createChildAlgorithm()` function or the `g_log` logger from the MDEventWorkspace file and this means that the solution cannot build - were you thinking of another way to do this?

SimonHeybrock, 2018-06-07:
For the logger, I think you can create one, see e.g., in `MatrixWorkspace.cpp`: `Kernel::Logger g_log("MatrixWorkspace");`.

Saving and reloading the data of the file backed workspace without algorithms may be more difficult. Maybe the functionality of `SaveMD` and `LoadMD` can be extracted into separate functions or classes, such that it can be used from the algorithms as well as from the workspace?

BhuvanBezawada, 2018-06-07:
Ah ok I see - the logger seems to work now. I'll have a look at `SaveMD` and `LoadMD` and see if I can extract any useful functionality. Thanks for your help!

closed 2021-03-02 by stale[bot]