mantidproject/mantid #13052: Engg (EnggUtils): take bank numbers from grouping file

FedeMPouzols, 2015-07-09:
At the moment this is being used for EnginX only and everythin is hardcoded. In EnggUtils, get the numbers 1, 2... from the Grouping file. The aliases 'Nort' and 'South' would be associated by default to 1 and 2 for EnginX.

It looks like later on we will be adding group 3 as a "texture" group in EnginX.

(Also: change that part of the code towards a more general treatment of detector groups, so that it is more or less ready for other instruments: the instrument should be guessed from the workspace whenever possible). Beware that the naming conventions of grouping files seem to be instrument dependent: ENGINX_Grouping.xml, CHRONUS_Detector_Grouping_LF_608.xml, etc.

FedeMPouzols, 2016-09-16:
@DavidFair, @DimitarTasev: this will be needed for other instruments to work well. We might get lucky because the next one in the queue (IMAT) will also have two diffraction banks. in principle 
But you see that the code is currently NOT doing the right thing. Much of the code in `EnggUtils` come from very early versions of the Engg system, keep your eyes open, as not all of `EnggUtils` needs to be in there but could be better integrated with the algorithms.

DimitarTasev, 2016-12-01:
Closed via https://github.com/mantidproject/mantid/issues/17999

closed 2016-12-01 by DimitarTasev