mantidproject/mantid #13077: Method selectTool() has no effect for the mask and pick tabs of the InstrumentVieew

FedeMPouzols, 2015-07-13:
Example script:
```
Load('164199.nxs', OutputWorkspace='example')
# HERE: right click on the 'example' workspace and open the instrument view

inst_view = getInstrumentView('example')
# these commands should change the tool in the 'Pick' tab but they don't work:
pick_tab = inst_view.getTab(InstrumentWindow.PICK)
pick_tab.selectTool(InstrumentWindowPickTab.PeakSelect)
pick_tab.selectTool(InstrumentWindowPickTab.Zoom)

# in the 'Mask' tab selectTool doesn't seem to work either:
mask_tab = inst_view.getTab(InstrumentWindow.MASK)
mask_tab.selectTool(InstrumentWindowMaskTab.DrawEllipse)
mask_tab.selectTool(InstrumentWindowMaskTab.DrawRectangle)
```

- In the last example of the wiki page, about the instrument tree tab, InstrumentWindowRenderTab.FULL3D is used, but the InstrumentWindowRenderTab class is not available in my command line:
```
InstrumentWindowRenderTab
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-44-ca8d5b94e3b3> in <module>()
----> 1 InstrumentWindowRenderTab

NameError: name 'InstrumentWindowRenderTab' is not defined
```
 I can only see InstrumentWindow, InstrumentWindowMaskTab and InstrumentWindowPickTab. It may be that InstrumentWindowRenderTab  is not exposed to Python in current versions or that there is something wrong in the way it is exposed and/or imported. This needs a bit of investigation if you're familiar with python and boost::python.

 This is part of / blocking #11766.

shahrozahmed1, 2015-08-04:
[image]
Seems to be fixed by #13220  and now recognises `InstrumentWindowRenderTab` and `InstrumentWindowRenderTab.FULL3D`

FedeMPouzols, 2015-08-04:
Just tested this also on debian after the changes in #13220. selectTool now runs correctly in principle, so I think we can close this issue as fixed elsewhere. But I just tried selectTool with the example instrument view and it doesn't really select the tools. We should check in the umbrella issue (#11766) if that's because of any other issue in the python interface of the instrument view and/or if it's possible to fix it.

closed 2015-08-04 by FedeMPouzols