mantidproject/mantid #13093: Function to read instrument parameter correction file

KarlPalmen, 2015-07-14:
This is step 5 of issue #12255.

It must be done after issue #13091 has been closed.

Write a function in LoadIDFFromNexus to read the instrument parameter correction file specified with full pathname and return/output the parameter file to use for correction or "" for no correction and whether the correction appends or replaces the embedded parameters.

Add unit test for this.

Optionally call it in the appropriate place in LoadIDFFromNexus to use the output of the function that finds the correction file if found. One may add if blocks in preparation of step 6.

KarlPalmen, 2015-07-22:
This is a duplicate of #13162 .

NickDraper, 2015-07-23:
duplicate agreed

closed 2015-07-23 by NickDraper