mantidproject/mantid #13096: MuonAnalysis improvements

raquelalvarezbanos, 2015-07-14:
James Lord has suggested the following improvements/enhancements:

> 1. Be able to overlay asymmetry plots of two ( or more) periods from the currently loaded run on the same graph.

> 2. More arbitrary arithmetic to combine periods, for example 1+3-2-4

> 3. Co-adding successive runs to improve stats (plain non period data too)

> 4. When changing period selection, at present it clears the fit curve off the graph. It would be nice to see how the old fit and new period compare. Similarly you could keep an old fit curve when loading the next run.

> 5. If fitting more than one period or combination from the same run, there's only one entry in the Results table. Should all of them be there, perhaps with an extra column for period number?

> 6. When adding two periods before analysis,  it should add the counts and then calculate the asymmetry. At present it appears to calculate asymmetries for each period and then add them. That makes it difficult to compare data when there are two Off periods and one On, for example.

The first thing to do is create different issues for each item in the list.

raquelalvarezbanos, 2015-07-15:
Issue #13102 has been created (item 6. in the list)

raquelalvarezbanos, 2015-10-16:
The interface already allows users to co-add successive runs. However when this is the case, the run information is not correct (issue #13738 was created to fix this).

tom-perkins, 2015-11-27:
Item 2 (arbitrary arithmetic to combine periods) is issue #14020, fixed in PR #14536

raquelalvarezbanos, 2015-11-27:
I've updated the description so items can be marked as fixed. I've marked some of them as resolved, but we have to have a meeting with scientists to discuss the remaining ones. Issue #13738 mentioned above does not need discussion.

tom-perkins, 2016-02-25:
Meeting held on 25/2/16.

- Item 9 (multi-dataset fitting) is a high priority, scientists will formulate use cases and we will circulate Balsamiq mockups for this
- One use case for this is doing fitting the same function to all periods in a run, perhaps with some shared parameters, and then using the sequential fit to repeat this for different runs. 
- Item 7 (tiled plots) is a popular request. It's useful to view data from all detectors, or all groups
- Item 1 (overlay) - Use a tickbox next to the "Plot" button to choose whether new plots are overlaid on the previous plots or not. It would be a commonly used option so put it here rather than in the Settings page.
- Item 5 - use the format `run number: period`, leave off the period for All or single-period data

tom-perkins, 2016-09-22:
Tiled plots: for fit results, this is possible by using the new feature in #17434 to generate such plots on right-clicking a WorkspaceGroup.
Still need the ability to do this for the data before it is fitted.

closed 2021-03-02 by stale[bot]