mantidproject/mantid #13107: Empty workspace title from LoadNexusProcessed (for some files)

FedeMPouzols, 2015-07-15:
Try this:
```
ews=LoadNexusProcessed('ENGINX_precalculated_vanadium_bank_curves')
ews.getTitle()
```
And you'll get an empty title, while the file has a title. The file and the title are generated correctly with SaveNexusProcessed, but it's then lost. There's this related (very old) issue #2201, and I seem to remember some other issue(s) related to get/setTitle which I cannot find now.

The file used in the example is being added in the unit test data in another branch. 

However, the title is found in other cases:
```
ews=LoadNexusProcessed('GEM38370_Focussed.nxs')
ews.getTitle()
# work well
```

Maybe it's because `ENGINX_precalculated_vanadium_bank_curves` does not have an instrument (it comes from a `Fit`)?

A quick look at the code took me to line 200 in LoadNexusProcessed.cpp, as they discuss in #2201, now around line 260.

FedeMPouzols, 2015-07-28:
This seems to be very closely related to #11386.

NickDraper, 2016-02-18:
Current warnings in the Rhel 7 clean master build are as follows

```
01:45:12 
01:45:12 /home/builder/Jenkins/workspace/master_clean-rhel7/docs/source/algorithms/LRAutoReduction-v1.rst:18: WARNING: undefined label: algm-liquidsreflectometryreduction (if the link has no caption the label must precede a section header)
01:45:12 /home/builder/Jenkins/workspace/master_clean-rhel7/docs/source/concepts/MatrixWorkspace.rst:372: WARNING: undefined label: plus (if the link has no caption the label must precede a section header)
01:45:12 /home/builder/Jenkins/workspace/master_clean-rhel7/docs/source/concepts/Properties.rst:179: WARNING: undefined label: algm-propertyalgorithm (if the link has no caption the label must precede a section header)
01:45:12 /home/builder/Jenkins/workspace/master_clean-rhel7/docs/source/concepts/WorkflowAlgorithm.rst:44: WARNING: undefined label: algm-muonload (if the link has no caption the label must precede a section header)
01:45:12 writing additional files... genindex py-modindex search
```

NickDraper, 2017-06-29:
duplicate of #11386

closed 2017-06-29 by NickDraper