mantidproject/mantid #13136: Solve various minor issues

wdzhou, 2015-07-18:
During the demo to the instrument scientist of HB2A, a couple of minor issues were raised to solve. 

1. GUI should be able to plot a specific detector’s counts vs. a sample log.  Error bar (mostly square root of the counts) should be plotted with data points; 
     Test case: Exp 243, Scan 16, Detector 11 vs. **sample_b**

2. There is some problem with zoom-in and zoom-out operation in the multiple-scans tab with plotting 1-D;

3. In the HB2A-standard detector exclusion file, detector ID starts from 0; while in the rest, it starts from 1.  Thus, it always has detector 43 got excluded, while 44 is the one to exclude. 

4. Enable the user to set the excluded detector manually and override the detector exclusion file; 

5. Plot all raw data directly from the file in multiple scan tab;

6. The color scheme of the lines should be reset with a new scan, and the light color should be removed from the list. 

For issues with tab-multiple scan, a good test case is Exp 416, Scan 10 to 20, delta(2theta) = 0.05

wdzhou, 2015-08-05:
After doing some tests, it is better to solve such problem in this issue.
1. GUI cannot plot 'raw' counts of detector in tab 'Individual detector'; 
2. Button 'Load Data' should be disabled if tab 3 (multi-scan) is chosen.

wdzhou, 2015-08-05:
Another issue is found with using error bar.  The returned from plot with error bar is an object of LineCollection  (for example, <matplotlib.collections.LineCollection object at 0x7f67fcf36110>,), which is not in axes.lines.

wdzhou, 2015-08-07:
In tab-2, the legend does not show correctly.  For example, detector 11 is plotted and then the image is cleared.  After detector 12 is plotted, the legend include detector 11.

wdzhou, 2015-08-07:
'Detectors Efficiency Correction' is applied at the stage of loading data (i.e., converting SPICE file to MDEventWorkspace).  Therefore, in order to enable or disable this feature, data file must be reloaded.

rosswhitfield, 2015-08-07:
Fixed in #13296

closed 2015-08-07 by rosswhitfield