mantidproject/mantid #13145: Improve Muonic XRD algorithms

NickDraper, 2015-07-20:
Here are two script written by one of our users Adrian Hiller that need to be improved.

``` python
from mantid.api import * # PythonAlgorithm, registerAlgorithm, WorkspaceProperty

def Muonic_XR(element):
    Muonic_XR=[['Au',8135.2,8090.6,8105.4,8069.4,5764.89,5594.97,3360.2,3206.8,2474.22,2341.21,2304.44,1436.05,1391.58,1104.9,899.14,869.98,405.654,400.143],
                        ['Ag',3184.7,3147.7,901.2,869.2,308.428,304.759],
                        ['Cu',1512.78,1506.61,334.8,330.26],
                        ['Zn',1600.15,1592.97,360.75,354.29],
                        ['Pb',8523.3,8442.11,5966.0,5780.1,2641.8,2499.7,2459.7,1511.63,1214.12,1028.83,972.3,938.4,437.687,431.285],
                        ['As',1866.9,1855.8,436.6,427.5],
                        ['Sn',3457.3,3412.8,1022.6,982.5,349.953,345.226]]

    i=None   
    for index, item in enumerate(Muonic_XR):
        if item[0] == element:
             i=index
    a=Muonic_XR[i][:] 
    return a

def Create_MuonicXR_WS(XR,YPos):

    YPos_WS=[YPos]*len(XR[1:])
    print type(YPos_WS)
    print type(XR[1:])
    X=(XR[1:])
    WS_Muon_XR=CreateWorkspace(X,YPos_WS[:])
    print type(X[1])
    RenameWorkspaces(WS_Muon_XR, WorkspaceNames=[XR[0]])
    return WS_Muon_XR    

class Get_NegMu_MuonicXRD(PythonAlgorithm):

	def PyInit(self):
                Element_type = ["Au", "Ag", "Cu","Zn","Pb","As","Sn"]
                self.declareProperty("Element", "None",
                             StringListValidator(Element_type),
                             "Select Element for Muonic XR")
                self.declareProperty(name="YAxis Position",defaultValue=-0.001,doc="Position for Markers on the y-axis")

	def category(self):
		return "Muon"

	def PyExec(self):
                ele=self.getProperty("Element").value
                ypos=self.getProperty("YAxis Position").value
                XR=Muonic_XR(ele)
                #print XR
                ws=Create_MuonicXR_WS(XR,ypos)

AlgorithmFactory.subscribe(Get_NegMu_MuonicXRD)
```
and 

``` python
from mantid.api import * # PythonAlgorithm, registerAlgorithm, WorkspaceProperty

def combine(dd,runno,A2000,B2000,A3000,B3000,C,C1,spec):
    if spec<10:
        s='0'+str(spec)
    else:
        s=str(spec)
    print dd+'ral0'+runno+'.rooth30'+s+'.dat'
    #loading data
    try:    
        ws3000=Load(Filename=dd+r'\ral0'+runno+'.rooth30'+s+'.dat', OutputWorkspace='ws3000')
        print 'hello'
        ws2000=Load(Filename=dd+r'\ral0'+runno+'.rooth20'+s+'.dat', OutputWorkspace='ws2000')
    except RuntimeError:
         print runno+' '+s+"not found"

    #xdata1=A2000*ws2000.readX(0)[:]+B2000

    #xdata4=A3000*ws3000.readX(0)[:]+B3000
   # print type(ws3000.readX(0)[:])
   #Correcting for Gain and offset of the detectors
    ws2000_corr=CreateWorkspace(A2000*ws2000.readX(0)[:]+B2000,ws2000.readY(0)[:])
    ws3000_corr=CreateWorkspace(A3000*ws3000.readX(0)[:]+B3000,ws3000.readY(0)[:])
 
    #Summing total counts for normalisation
    ws2000_total=0
    ws3000_total=0
    for x in range (0,8000):
        try:
            ws2000_total=ws2000_corr.readY(0)[x]+ws2000_total
            ws3000_total=ws3000_corr.readY(0)[x]+ws3000_total
        except:
             #print 'hello'
             continue
             
    print ws2000_total
    print ws3000_total
    #normalising
    ws2000_corr=ws2000_corr/ws2000_total
    ws3000_corr=ws3000_corr/ws3000_total
    
    #rebinning to add detectors together
    bin=[100,ws2000.readX(0)[2]-ws2000.readX(0)[1],8000]
    
    ws2000_corr_rebin=Rebin(ws2000_corr,bin)
    ws3000_corr_rebin=Rebin(ws3000_corr,bin)

    ws_ral=Plus(ws2000_corr_rebin,ws3000_corr_rebin)
    
    suf='_'+str(spec)+'_'+runno
    
    RenameWorkspaces(ws_ral,Suffix=suf)
    RenameWorkspaces(ws2000_corr,Suffix=suf)
    RenameWorkspaces(ws3000_corr,Suffix=suf)
    
  #  mantidplot.plot_spectrum(ws_ral,0)
    DeleteWorkspace(ws2000)
    DeleteWorkspace(ws3000)
  #  DeleteWorkspace(ws2000_corr)
   # DeleteWorkspace(ws3000_corr)
    DeleteWorkspace(ws2000_corr_rebin)
    DeleteWorkspace(ws3000_corr_rebin)

    return

class ApplyNegMuCorrection(PythonAlgorithm):

	def PyInit(self):
		self.declareProperty(name="Data Directory",defaultValue=r'M:\Data\Negative Muons\forMantid',doc="Data directory")
		self.declareProperty(name="First Run Number",defaultValue=1718,doc="First Run Number")
		self.declareProperty(name="Last Run Number",defaultValue=1719,doc="Last Run Number")
		self.declareProperty(name="Gain RIKEN High E",defaultValue=1.077,doc="Gain RIKEN High E")
		self.declareProperty(name="Offset RIKEN High E",defaultValue=-1,doc="OffSet RIKEN High E")
		self.declareProperty(name="Gain ISIS High E",defaultValue=1.278,doc="Gain ISIS High E")
		self.declareProperty(name="Offset ISIS High E",defaultValue=-12,doc="OffSet ISIS High E")
		self.declareProperty(name="Gain ISIS Low E",defaultValue=1.2,doc="Gain ISIS Low E")
		self.declareProperty(name="Offset ISIS Low E",defaultValue=0.0,doc="OffSet ISIS Low E")
 
	def category(self):
		return "CorrectionFunctions;Muon"

	def PyExec(self):
        
                spec=1
                dd = self.getProperty("Data Directory").value
                first = self.getProperty("First Run Number").value
                last = self.getProperty("Last Run Number").value+1
                GRHE=self.getProperty("Gain RIKEN High E").value
                ORHE=self.getProperty("Offset RIKEN High E").value
                GIHE=self.getProperty("Gain ISIS High E").value
                OIHE=self.getProperty("Offset ISIS High E").value
                GILE=self.getProperty("Gain ISIS Low E").value
                OILE=self.getProperty("Offset ISIS Low E").value
        
                for x in range(first,last):
                    for spec in range(0,3):
                        runno=str(x)
                        combine(dd,runno,GRHE,ORHE,GIHE,OIHE,GILE,OILE,spec)
                    combine(dd,runno,GRHE,ORHE,GIHE,OIHE,GILE,OILE,10)

		

AlgorithmFactory.subscribe(ApplyNegMuCorrection)
```

Both need improving to support our python algorithm way of doing things, and  the first one should be extended to allow either a single element to be chosen,or a list of strings of the elements to be entered, (in which case it should return a workspace group.

MattKing06, 2015-08-07:
This issue has been split into 3 separate issues: #13155 (GetNegMuMuonicXRD Algorithm) #13146 (GUI for GetNegMuMuonicAlgorithm) and #13201 (ApplyNegMuCorrection Algorithm)

closed 2015-08-07 by FedeMPouzols

FedeMPouzols, 2015-08-07:
@MattKing06 : in case you're not familiar with it. You can use check lists in GitHub issues when you have a "meta" or "umbrella" issue like this one which depends on 3. See for example: https://github.com/mantidproject/mantid/issues/9213. That way you can keep track of the three child issues, tick them when they're done and finally when you tick them all you would set this parent issue as "Review Required" saying something like "this issue was addressed in the three children, please check that the initial requirements have been fulfilled".

Oh, actually it seems to me that in addition to the #13155, #13146, and #13201, this parent issue has also spawned #13145, #13333, and #13334. Is that correct? Then there would be 6 issues in the check list.

you should also add here the three issues related to

PR #13155 (2015-08-07): Improve GetNuMegMuonicXRD Algorithm
fixes #13145

Implemented a dictionary to store elements and peak
values instead of an array of arrays. Updated code
to comply with data structure use

Refs #13145