mantidproject/mantid #13157: Create PDF from rst for training pages

NickDraper, 2015-07-21:
The aim here was to mimic the PDF creation functionality of mediawiki to allow the training courses to be fully migrated over to rst pages.

This excercise has now been abandoned, but is documented here to capture the attempt.

NickDraper, 2015-07-21:
Fundamental requirements of the outputed PDF:
1. well formatted text and headings
1. properly positioned and sized images
1. text wrapping around right aligned images
1. good formatting of code samples
1. good formatting of tables (including embedded images)

NickDraper, 2015-07-21:
Attempt using Latex builder in Sphinx, then pdf2latex:

Problems:
1. Image sizing is not preserved
1. Lots of blank pages before every heading, leading to Mantid basics course being 160 rather than about 90 pages.
1. Image positioning hard to tell as all images too large
1. Long table text overruns columns.
1. Code samples ok but not great

NickDraper, 2015-07-21:
Attempt using rst2pdf builder in sphinx:

Problems:
1. installing reportlab is a right royal pain
1. Poor choice of where to put page breaks, often straight after section headings
1. Inline images overlay text, obscuring it
1. Text wrapping of right aligned images is not possible (apparently a limitation in Reportlab)

NickDraper, 2015-07-21:
Return to Latex

Attempts to solve problems

1. Image sizing is preserved IF the size is defined in em, rather than px
1. The Page breaks can be suppressed using some clever preabmle hack, however the format of the chapter titles are too over done for this style of docuemnt.

 ```
\usepackage{etoolbox}
\makeatletter
\patchcmd{\chapter}{\if@openright\cleardoublepage\else\clearpage\fi}{}{}{}
\makeatother
 ```
[image]

1. Images can have text wrapping, but only as a post Sphinx edit of the tex file. eg.
 ```
Sphinx right aligned image declaration:
{\hfill\includegraphics{ShowLoadandWorkspaceAreaInMantidPlot.PNG}}

Needs to be edited to:
\begin{wrapfigure}{r}{19em}
\includegraphics[width=18.0em]{ManageDirectories.png}
\end{wrapfigure}
 ```
 and this can work.
 [image]
 However:

 1. This needs to be done as an additional post processing step, written by us
 1. The size of the wrap area has to be defined:
   1. This is not too hard as long as the image has a size directive
   1. but is much harder if it is not defined (which is frequently the case).
 1. While I haven't attempted this it is possible that the Long table text overruns columns could be resolved using substitutions for links and images to reduce the relative table column width in the rst.

NickDraper, 2015-07-21:
Given the problems and the level of manual intervention needed to overcome these I have decided to abandon this effort and continue with the training courses in mediawiki for now.

NickDraper, 2015-07-21:
Marked as wontfix as effort abandoned with great dissapointment

FedeMPouzols, 2015-09-24:
Big disappointment. The rst/latex/pdf tools are definitely not able to produce quality pdf output. The steps and issues found are well documented here.

closed 2015-09-24 by FedeMPouzols