mantidproject/mantid #13162: Function to read Parameter Correction File and output File Name and Append flag if applicable.

KarlPalmen, 2015-07-21:
This is step 5 of issue #12255.

The function will take the correction file and date as input arguments and output the name of the parameter correction file and append flag if applicable.

The function shall be called from LoadIDFFromNexus using the parameter correction file found by it and the date obtained from the getWorkspaceStartDate function of the workspace. It will then log the outputs at debug level.

Also add unit test for it. Hopefully ScopedFiles can be used as test files for this unit test.

KarlPalmen, 2015-07-22:
The suggested format of the XML parameters correction file is

 \<embedded-parameter-corrections name=”XXX”>
     \<correction  valid-from=””  valid-to=”” file=”filename” append=’false’/>
.
.   
.
  \<correction  valid-from=””  valid-to=”” file=”filename” append=’true’/>
\</ embedded-parameter-corrections>

with appropriate indentation, which I haven't figured how to show. Nor do I know the equivalent to trac '{{{'. 

To keep implementation simple and simple to test only one embedded-parameter-corrections item will be read and its name attribute will be ignored.

stuartcampbell, 2015-07-22:
@KarlPalmen If you need help with markdown syntax then there is a button labeled "Markdown supported" just about the input text field that will tell you most things.  But for this case {{{ == ```

KarlPalmen, 2015-07-22:
Dear Stuart

Thank you.

Karl

closed 2015-07-27 by KarlPalmen