mantidproject/mantid #13164: Convert image sizes to em in help pages

NickDraper, 2015-07-21:
images sized in em scale much better across different screen sizes.

px sizing is artificial based on a particular persons screen and will look different at different resolutions.

NickDraper, 2015-07-21:
Once this was done, it was possible to see that this is not necessary in modern browsers, it makes no difference to IE or chrome.  In QTAssistant we have the zooming functionality disabled.

This would be important if we ever resurrected conversion to pdf via latex, but otherwise it makes no difference.  Also on the whole developers understand (partially) what a pixel is and means, but few understand an em.

Therefore this will not go forward and is marked as won't fis, the branch will be deleted, but the code is here for reference.

FedeMPouzols, 2015-09-24:
Could become relevant again in the future for rst => latex/pdf conversion (#13157).

closed 2015-09-24 by FedeMPouzols