mantidproject/mantid #13171: SaveNXSPE does not produces correct detectors for MAPS, rings map

abuts, 2015-07-22:


AndreiSavici, 2015-07-22:
For powder pattern you should use ```GenerateGroupingPowder```. It produces a map file and a par file. Use the par file in ```SaveNXSPE```

abuts, 2015-07-22:
We are investigating -- the problem apparently not in SaveNXSPE -- something to to with 4-1 map changes. Need to identify when and why it have happened. Couple of experiments clearly need to be reprocessed.
I am taking this ticket and it will probably be invalid.

abuts, 2015-07-23:
We investigated the issue and realized that it arises due to incorrect ring map (built on 4-to-1 processed file rather then the file, obtained from instrument). As such the issue have nothing to do with Mantid.

closed 2015-07-23 by abuts