mantidproject/mantid #13198: Can't change attribute values in python functions

raquelalvarezbanos, 2015-07-24:
James Lord reported:

> Try the simple function below. Suitable data would be two spectra with oscillations 90 degrees out of phase. I find that numeric Attributes are now fixed to the initial values specified in the code and I can't change them - if I type a new value it resets to the original. 

> My code closely resembles that on the help pages http://www.mantidproject.org/Python_Function_Attributes. Is this wrong?

```
import numpy
import math
from mantid.api import *
from mantid.kernel import *
from mantid.simpleapi import *
numpy.set_printoptions(linewidth=100)

class AttributeExample(IFunction1D):
	# model "resonance wiggles"
	# integrated linear background function
	def init(self):
		self.declareParameter("Amplitude",0.2)
		self.declareParameter("Baseline",0.1)
		self.declareAttribute("Frequency",0.26)
		self.declareAttribute("Sine","False")

# not used any more?
	def setAttributeValue(self,name,value):
		if name == "Frequency":
			self._freq = value
		if name == "Sine":
			self._sine = value
			
	def function1D(self,xvals):
		ampl=self.getParameterValue("Amplitude")
		base=self.getParameterValue("Baseline")
		
		#self._nspec=self.getAttributeValue("Nspec")
		#self._y0=self.getAttributeValue("Y0")
		#self._dY=self.getAttributeValue("dY")
		
		if(self._sine):
			ybins=numpy.sin(xvals*self._freq*2.0*math.pi)*ampl+base
		else:
			ybins=numpy.cos(xvals*self._freq*2.0*math.pi)*ampl+base
		return ybins
		
FunctionFactory.subscribe(AttributeExample)
```

raquelalvarezbanos, 2015-07-24:
More from James Lord:

> Another future improvement: some of Mantid's built in functions such as Polynomial and Chebyshev use Attributes to determine how many parameters will be provided. It would be useful to let Python functions do this as well: at present I have a series of functions QuantumTableDrivenFunction1, QuantumTableDrivenFunction2, QuantumTableDrivenFunction3, ... according to how many fit variables are needed for a specific model

LamarMoore, 2015-12-15:
The reason why this doesn't work is because there is a missing call to the super class which will actually set and update the attritbutes.

``` python
super(AttributeExample, self).setAttributeValue(name, value)
```

This addition should allow one to set/store the attribute values correctly.

closed 2015-12-17 by LamarMoore