mantidproject/mantid #13215: IPython script interpreter: autocompletion crashes

FedeMPouzols, 2015-07-27:
Start writing for example `InstrumentW` and then use the TAB-autocompletion. It works but it immediately crashes on an assert (QtCored4.dll, qglobal.cpp:2303, "f == fname" in text\qtextdocument_p.cpp: 1500).

This is the trace that you get If you do Retry -> Debug:
```
 	QtCored4.dll!qt_message_output(QtMsgType msgType, const char * buf) Line 2311	C++
 	QtCored4.dll!qt_message(QtMsgType msgType, const char * msg, char * ap) Line 2369	C++
 	QtCored4.dll!qFatal(const char * msg, ...) Line 2553	C++
 	QtCored4.dll!qt_assert(const char * assertion, const char * file, int line) Line 2019	C++
 	QtGuid4.dll!QTextDocumentPrivate::scan_frames(int pos, int charsRemoved, int charsAdded) Line 1500	C++
 	QtGuid4.dll!QTextDocumentPrivate::finishEdit() Line 1214	C++
 	QtGuid4.dll!QTextDocumentPrivate::endEditBlock() Line 1202	C++
 	QtGuid4.dll!QTextDocumentFragmentPrivate::QTextDocumentFragmentPrivate(const QTextCursor & _cursor) Line 235	C++
 	QtGuid4.dll!QTextDocumentFragment::QTextDocumentFragment(const QTextCursor & cursor) Line 314	C++
 	QtGuid4.dll!QTextCursor::selection() Line 1770	C++
 	QtGui_d.pyd!000007fec1dae97a()	Unknown
 	python27_d.dll!000000001e17b438()	Unknown
 	python27_d.dll!000000001e2178a4()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e15b73d()	Unknown
 	python27_d.dll!000000001e109abb()	Unknown
 	python27_d.dll!000000001e218bce()	Unknown
 	python27_d.dll!000000001e2153d0()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e218e5a()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e218dbf()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e218dbf()	Unknown
 	python27_d.dll!000000001e217989()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e15b73d()	Unknown
 	python27_d.dll!000000001e109abb()	Unknown
 	python27_d.dll!000000001e129169()	Unknown
 	python27_d.dll!000000001e109abb()	Unknown
 	python27_d.dll!000000001e20f3c6()	Unknown
 	sip_d.pyd!000007fecb3abe92()	Unknown
 	QtCore_d.pyd!000007fec2b45ebb()	Unknown
 	QtCore_d.pyd!000007fec2b45c0b()	Unknown
 	QtCore_d.pyd!000007fec2b45b07()	Unknown
 	QtCored4.dll!QMetaObject::metacall(QObject * object, QMetaObject::Call cl, int idx, void * * argv) Line 246	C++
 	QtCored4.dll!QMetaObject::activate(QObject * sender, const QMetaObject * m, int local_signal_index, void * * argv) Line 3558	C++
 	QtCored4.dll!QMetaObject::activate(QObject * sender, int signal_index, void * * argv) Line 3616	C++
 	QtCored4.dll!QMetaObject::activate(QObject * sender, int from_signal_index, int to_signal_index, void * * argv) Line 3435	C++
 	QtCore_d.pyd!000007fec2b34bcb()	Unknown
 	QtCore_d.pyd!000007fec2b39971()	Unknown
 	python27_d.dll!000000001e17b438()	Unknown
 	python27_d.dll!000000001e2178a4()	Unknown
 	python27_d.dll!000000001e215283()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e15b73d()	Unknown
 	python27_d.dll!000000001e109abb()	Unknown
 	python27_d.dll!000000001e218bce()	Unknown
 	python27_d.dll!000000001e2153d0()	Unknown
 	python27_d.dll!000000001e20fdaa()	Unknown
 	python27_d.dll!000000001e15b73d()	Unknown
 	python27_d.dll!000000001e109abb()	Unknown
 	python27_d.dll!000000001e20f3c6()	Unknown
 	sip_d.pyd!000007fecb3abe92()	Unknown
 	QtCore_d.pyd!000007fec2b45ebb()	Unknown
 	QtCore_d.pyd!000007fec2b45c0b()	Unknown
 	QtCore_d.pyd!000007fec2b45b07()	Unknown
 	QtCored4.dll!QMetaObject::metacall(QObject * object, QMetaObject::Call cl, int idx, void * * argv) Line 246	C++
 	QtCored4.dll!QMetaCallEvent::placeMetaCall(QObject * object) Line 528	C++
 	QtCored4.dll!QObject::event(QEvent * e) Line 1194	C++
 	QtGuid4.dll!QApplicationPrivate::notify_helper(QObject * receiver, QEvent * e) Line 4562	C++
 	QtGuid4.dll!QApplication::notify(QObject * receiver, QEvent * e) Line 3944	C++
>	MantidPlot.exe!MantidApplication::notify(QObject * receiver, QEvent * event) Line 29	C++
 	QtCored4.dll!QCoreApplication::notifyInternal(QObject * receiver, QEvent * event) Line 946	C++
 	QtCored4.dll!QCoreApplication::sendEvent(QObject * receiver, QEvent * event) Line 231	C++
 	QtCored4.dll!QCoreApplicationPrivate::sendPostedEvents(QObject * receiver, int event_type, QThreadData * data) Line 1570	C++
 	QtCored4.dll!qt_internal_proc(HWND__ * hwnd, unsigned int message, unsigned __int64 wp, __int64 lp) Line 498	C++
 	user32.dll!0000000077769bd1()	Unknown
 	user32.dll!00000000777698da()	Unknown
 	QtCored4.dll!QEventDispatcherWin32::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 810	C++
 	QtGuid4.dll!QGuiEventDispatcherWin32::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 1204	C++
 	QtCored4.dll!QEventLoop::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 150	C++
 	QtCored4.dll!QEventLoop::exec(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 204	C++
 	QtCored4.dll!QCoreApplication::exec() Line 1218	C++
 	QtGuid4.dll!QApplication::exec() Line 3824	C++
 	MantidPlot.exe!main(int argc, char * * argv) Line 215	C++
 	MantidPlot.exe!__tmainCRTStartup() Line 536	C
 	MantidPlot.exe!mainCRTStartup() Line 377	C
 	kernel32.dll!00000000775c59cd()	Unknown
 	ntdll.dll!000000007787b891()	Unknown
```

martyngigg, 2015-08-03:
This only happens with a debug build so is not so high priority. It looks like upgrading IPython solves the problem and that is most likely to happen when we move to VS 2015

martyngigg, 2015-11-19:
@FedeMPouzols Does this still happen now?

FedeMPouzols, 2015-11-19:
@martyngigg : yes. I just got this crash on a fresh build and I remember we got it with Matt a couple of days ago. The call stack looks impenetrable:

```
>	QtCored4.dll!0000000066c2b243()	Unknown
 	QtCored4.dll!0000000066c2c032()	Unknown
 	QtCored4.dll!0000000066c2acbb()	Unknown
 	QtCored4.dll!0000000066c2b00e()	Unknown
 	QtGuid4.dll!0000000064c3df4c()	Unknown
 	QtGuid4.dll!0000000064c3b28c()	Unknown
 	QtGuid4.dll!0000000064c3b1f3()	Unknown
 	QtGuid4.dll!0000000064c84c34()	Unknown
 	QtGuid4.dll!0000000064c832bc()	Unknown
 	QtGuid4.dll!0000000064c7a9e5()	Unknown
 	QtGui.pyd!000007fec8041faa()	Unknown
 	python27.dll!000000001e0c3d49()	Unknown
 	python27.dll!000000001e1146d4()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e0b5833()	Unknown
 	python27.dll!000000001e08dae5()	Unknown
 	python27.dll!000000001e1141cd()	Unknown
 	python27.dll!000000001e117fbf()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e112df9()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e112d88()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e112d88()	Unknown
 	python27.dll!000000001e114764()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e0b5833()	Unknown
 	python27.dll!000000001e08dae5()	Unknown
 	python27.dll!000000001e09bfd1()	Unknown
 	python27.dll!000000001e08dae5()	Unknown
 	python27.dll!000000001e112b88()	Unknown
 	sip.pyd!000007feea3cdbd1()	Unknown
 	QtCore.pyd!000007fec8df3262()	Unknown
 	QtCore.pyd!000007fec8df3052()	Unknown
 	QtCore.pyd!000007fec8df2f3a()	Unknown
 	QtCored4.dll!0000000066e18bd4()	Unknown
 	QtCored4.dll!0000000066e2d1a0()	Unknown
 	QtCored4.dll!0000000066e2ca20()	Unknown
 	QtCored4.dll!0000000066e2ca5a()	Unknown
 	QtCore.pyd!000007fec8ddadae()	Unknown
 	QtCore.pyd!000007fec8de24f0()	Unknown
 	python27.dll!000000001e0c3d49()	Unknown
 	python27.dll!000000001e1146d4()	Unknown
 	python27.dll!000000001e117ee4()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e0b5833()	Unknown
 	python27.dll!000000001e08dae5()	Unknown
 	python27.dll!000000001e1141cd()	Unknown
 	python27.dll!000000001e117fbf()	Unknown
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e0b5833()	Unknown
 	python27.dll!000000001e08dae5()	Unknown
 	python27.dll!000000001e112b88()	Unknown
 	sip.pyd!000007feea3cdbd1()	Unknown
 	QtCore.pyd!000007fec8df3262()	Unknown
 	QtCore.pyd!000007fec8df3052()	Unknown
 	QtCore.pyd!000007fec8df2f3a()	Unknown
 	QtCored4.dll!0000000066e18bd4()	Unknown
 	QtCored4.dll!0000000066e3452c()	Unknown
 	QtCored4.dll!0000000066e2eb86()	Unknown
 	QtGuid4.dll!00000000647b17b6()	Unknown
 	QtGuid4.dll!00000000647abfb8()	Unknown
 	MantidPlot.exe!MantidApplication::notify(QObject * receiver, QEvent * event) Line 30	C++
 	QtCored4.dll!0000000066e0b92a()	Unknown
 	QtCored4.dll!0000000066e14903()	Unknown
 	QtCored4.dll!0000000066e0cab6()	Unknown
 	QtCored4.dll!0000000066e61f14()	Unknown
 	[External Code]	
 	QtCored4.dll!0000000066e629c8()	Unknown
 	QtGuid4.dll!00000000648a98a5()	Unknown
 	QtCored4.dll!0000000066e07474()	Unknown
 	QtCored4.dll!0000000066e0774e()	Unknown
 	QtCored4.dll!0000000066e097dc()	Unknown
 	QtGuid4.dll!00000000647abb98()	Unknown
 	MantidPlot.exe!main(int argc, char * * argv) Line 226	C++
 	[External Code]	
```

martyngigg, 2015-11-19:
I'll have a look at upgrading IPython to 3.x soon, given the questions surrounding IPython 4.0/Jupyter - #13481

EltayebAhmed, 2016-08-15:
@martyngigg 
I encountered this issue with a current build of master, which is using Ipython 3.2.3

martyngigg, 2016-08-15:
@EltayebAhmed Are you using a debug build? This a known issue at the moment. Can you use a release build?

martyngigg, 2019-07-15:
This only occurs with Qt4 in MantidPlot and is not worth the effort of fixing.

closed 2019-07-15 by martyngigg