mantidproject/mantid #13218: Make SmoothMD work from the SliceViewer

OwenArnold, 2015-07-27:
* Familiarise yourself with [SmoothMD](http://docs.mantidproject.org/nightly/algorithms/SmoothMD-v1.html) (see algorithm documentation)
* Familiarise yourself with the (SliceViewer)[http://www.mantidproject.org/MantidPlot:_SliceViewer] (see wiki sliceviewer documentation)

** For any viewed slice. We ought to be able to apply SmoothMD in the SliceViewer. The aim of this issue is to generate a brief design and balsamiq mock-up for what this would look like. You also ought to think about how this might work. Investigate the existing Rebinning and LineOverlay modes in the SliceViewer and think about how you might code this**   We can review both documents together prior to commencing coding.

Things to consider:
* SmoothMD allows you to specify the width in each dimension. Maybe we could introduce this as a number entry next to each of the dimension sliders
* We should be able to switch the smoothing on/off, like you can switch the rebinnin on/off
* You should only be able to apply SmoothMD on an MDHistoWorkspace, so if you're using the SliceViewer on anything else, smoothing should not be available.
* If you have entered the interactive Rebin mode, you should also be able to Smooth (because you have a temporary MDHistoWorkspace) even if the workspace you loaded was not a MDHistoWorkspace.

OwenArnold, 2015-07-27:
Ask @picatess for help as you go

MattKing06, 2015-08-07:
Below is a Mockup of what SliceViewer might look like with this functionality added and also a Document explaining some of the basic logic involved with SmoothMD and how it can be used with the workspaces provided for SliceViewer.

[SliceViewerWithSmoothMDMockUp.pdf](https://github.com/mantidproject/mantid/files/1233/SliceViewerWithSmoothMDMockUp.pdf)
[SmoothMDinSliceViewer_WriteUp.pdf](https://github.com/mantidproject/mantid/files/1234/SmoothMDinSliceViewer_WriteUp.pdf)

OwenArnold, 2015-08-07:
This looks pretty good, here are my comments:

* Smoothing should result in a different output workspace, which should appear in the Analysis Data Service (ask me about this if unclear) and remain there until removed.
* Defaulting the Width parameters to 3 is a sensible default. Width Parameter boxes should only appear when the Smooth mode is selected, and disappear when removed.
* Technically you should be able to keep rebin and smooth on at the same time.
* Simple smooth is a re-appy operation. We should call it that in the documents

MattKing06, 2015-08-07:
Updated versions of Mockup and Writeup after our discussion. To see how the Mock-up will work view it in full-screen and the SmoothMD toggle and simple smooth button should work as intended.

[SliceViewerWithSmoothMDMockUp.pdf](https://github.com/mantidproject/mantid/files/1235/SliceViewerWithSmoothMDMockUp.pdf)
[SmoothMDinSliceViewer_WriteUp.pdf](https://github.com/mantidproject/mantid/files/1236/SmoothMDinSliceViewer_WriteUp.pdf)

MattKing06, 2015-08-10:
The Write-up and Mock-up of SmoothMD in Sliceviewer can be accessed from:
https://github.com/mantidproject/documents/tree/master/Design/VATES/Smoothing

OwenArnold, 2015-08-10:
* Re-write this ticket description to make clear that this is the mock-up and design phase. The outputs should only be design documents.
* In your write up, you refer to some detailed member variables in the code. Remove this, that doesn't live in this section. 
* In the mockups, you still refer to 'Simple Smooth' in places.

That should be it.

closed 2015-09-14 by NickDraper