mantidproject/mantid #13229: Filter Events Broken

shahrozahmed1, 2015-07-28:
if you go to Interface, Utility, Filter events the interface doesnt seem to work and instead displays this:
```
ImportError: No module named _path
  at line 2 in '<Interface>'
  caused by line 4 in 'C:/Users/RYQ25391/Documents/GitHub/mantid/Code/Mantid/scripts\FilterEvents\eventFilterGUI.py'
  caused by line 546 in 'C:/Users/RYQ25391/Documents/GitHub/mantid/Code/Mantid/scripts\FilterEvents\ui_MainWindow.py'
  caused by line 5 in 'C:/Users/RYQ25391/Documents/GitHub/mantid/Code/Mantid/scripts\FilterEvents\MplFigureCanvas.py'
  caused by line 15 in 'C:\Users\RYQ25391\Documents\MantidBuild\bin\Debug\lib\site-packages\matplotlib\backends\backend_qt4agg.py'
  caused by line 28 in 'C:\Users\RYQ25391\Documents\MantidBuild\bin\Debug\lib\site-packages\matplotlib\figure.py'
  caused by line 12 in 'C:\Users\RYQ25391\Documents\MantidBuild\bin\Debug\lib\site-packages\matplotlib\artist.py'
  caused by line 39 in 'C:\Users\RYQ25391\Documents\MantidBuild\bin\Debug\lib\site-packages\matplotlib\transforms.py' ```

shahrozahmed1, 2015-08-05:
These errors are only displayed on Windows debug mode and seems work in the release mode.

NickDraper, 2016-02-19:
Just tested on windows in debug mode, it must have been fixed

Works4me

closed 2016-03-15 by NickDraper