mantidproject/mantid #13258: Missing ParameterMap copy constructor

SimonHeybrock, 2015-07-30:
`ParameterMap` holds pointers to `Parameter` in a `map`. There is no copy constructor, so this will cause a shallow copy.

This invalidates the copy-on-write design when accessing `ParameterMap`.

There are no actual issues I know of that are related to this bug. This may indicate that the whole design is more complex than it needs to be? I am considering a redesign of this anyways, in the context of improving performance.

NickDraper, 2016-02-12:
This class is likely to be removed as part of the geomerty redesign effort

SimonHeybrock, 2016-02-12:
@NickDraper It is unlikely that the existing instrument code will be removed entirely any time soon. Regardless of that, open tickets are an important means for keeping track of bugs, and in this case also what to avoid in the new design. Having this closed means that it will not show up in searches by default, and will thus be forgotten.

Please either reopen this ticket or provide other reliable and searchable tools of tracking bugs. Consider creating a new label `will-not-fix`.

NickDraper, 2016-02-12:
That's what the archived label effectively is, although it covers a few other situations as well.

Archived = we are not going to do anything about this for now.

SimonHeybrock, 2016-02-14:
Ok, I am fine with that label. My point is that when it is *closed*, it does not show up in searches unless you explicitly search for closed issues, which, I guess, few people do.

closed 2021-03-02 by stale[bot]