mantidproject/mantid #13299: Try to parallelize Fit - investigate

FedeMPouzols, 2015-08-05:
I was fitting a BSpline of order 40 which took ~1m30s, and noticed that it used just one core. Users would appreciate a lot a speed up which in principle could be easy to get.

Investigate if Fit behaves like this always, and see where it would be a good place to use open mp directives, if possible. Not sure if GSL etc. code involved in this are reentrant/safe.

FedeMPouzols, 2015-08-05:
Note: BSpline of order 40 for the first spectrum of `ENGINX_precalculated_vanadium_run000236516_bank_curves.nxs` (found in unit and doc tests data).

martyngigg, 2015-08-05:
There is a lot to this and I think @mantid-roman has looked at it before. Was it a fit over a single spectrum that was slow or over the whole thing as a series of fits over all spectra? It may be a better bet to parallelize and run fits on separate spectra in parallel?

FedeMPouzols, 2015-08-05:
In this case it was a fit over a single spectrum (sum of spectra of a bank).
I thought one alternative could be to parallelize function evaluations over the x values. But that may clash with something else, and there may be issues with gsl, etc.
Parallelization over multiple spectra sounds like a good alternative, as that's the kind of use case that would need/benefit more from parallel loops. But that would be algorithm/technique specific? There are a few source files under `CurveFitting` that use parallel loops but they seem to be specific algorithms like `ConvertToYSpace` , `ConvolveWorkspaces` etc.
ENGIN-X scientists apparently do manual fits on one or a few spectra, go for a coffee, etc. and once they are happy with the way it works then they would run a batch fit over multiple spectra and/or workspaces. So if we can find a way to speed up single fits that would be great.

mantid-roman, 2015-08-05:
Parallelising function evaluation at eg IFunction1D level is the simplest way to do it. It shouldn't clash with anything apart from parallel loops inside concrete functions.

At the level of minimizers it also can be done (see ParDomain) but it isn't working with numerical derivatives at the moment.

closed 2021-03-01 by stale[bot]