mantidproject/mantid #13309: LoadVesuvio cannot be loaded by MantidPlot

martyngigg, 2015-08-05:
Starting MantidPlot now gives me a warning stating:
```
Failed to load plugin /mnt/data1/source/github/mantidproject/mantid/Code/Mantid/Framework/PythonInterface/plugins/algorithms/LoadVesuvio.py. Error: No module named LoadEmptyVesuvio```

It's not clear why it can't find it as it is in the same directory.

DanNixon, 2015-08-05:
On what platform, there is no error on Ubuntu for me.

martyngigg, 2015-08-05:
This was Ubuntu too - maybe it's my setup. I'll look a little closer.

DanNixon, 2015-08-05:
But I do get it on RHEL 6.

DanNixon, 2015-08-05:
I can't see why this is happening. The easiest thing I can think of that will fix it is adding the path to the algorithm directories to the Python path when Mantid is launched.

martyngigg, 2015-08-05:
I can't either. It should be adding the directory already while it starts up so that they are all importable

DanNixon, 2015-08-05:
I'm starting to wonder if this is caused by something else, as the systemtests for the PR and nightly since you passed my PR making the change to `LoadVesuvio` have passed until 03/08, now the Vesuvio tests all fail with the same error.

martyngigg, 2015-08-05:
Yes, that is odd. Something must have affected the path at start up

martyngigg, 2015-08-05:
Looking closer I'm not sure the path is now affected on startup. We use the `imp` module to load the filepath directly to avoid the need to search. This may be the problem. Although it doesn't explain why it did work...

DanNixon, 2015-08-05:
I'm guessing the difference between 2.6 and 2.7 somehow, since the only platform this seems to fail on (at least according to the systemtests) is RHEL6.

martyngigg, 2015-08-05:
I'm on Ubuntu though and I still see the same issue

martyngigg, 2015-08-05:
This is strange. On my machine the order that the python plugins are loaded is different to the Ubuntu build server even though the are exactly the same versions of Python. On the builder `LoadEmptyVesuvio` is loaded first and then when `LoadVesuvio` comes around it already has `LoadEmptyVesuvio` in `sys.modules` so there is nothing to do.

On my machine and the RHEL6 builder `LoadVesuvio` is loaded first and fails to find `LoadEmptyVesuvio` because the plugin directory is not on the path. There's no real reason why the plugin directory is not added to the path so I'll do that.

PR #13317 (2015-08-06): Add the Python plugin directories to sys.path
Fixes #13309

No release note updates are required.

**Tester**
Some systems load the Python plugins in a different order so their directories need to be on `sys.path` or they cannot find any dependent modules in the same directory. All tests should pass and a code review should be sufficient.
closed 2015-08-06 by DanNixon