mantidproject/mantid #13348: Add Fit systemtest

Anders-Markvardsen, 2015-08-10:
Add Fit systemtest which test a real fitting example

MichaelWedel, 2015-08-10:
When I added the unweighted least squares cost function Roman pointed out to me that NIST has a kind of test suite for non-linear least squares codes. I implemented some of them in `UnweightedLeastSquaresTest.py`, checking values and errors. Maybe more of the NIST examples could be added as well?

Anders-Markvardsen, 2015-08-17:
@MichaelWedel thanks for pointing this out

Anders-Markvardsen, 2015-08-17:
Looking at this in more detail there are in addition to UnweightedLeastSquaresTest.py many instrument specific systemtests using fitting. As expected most of these are not calling Fit directly (an exception being VesuvioFittingTest.py). 

A suggestion is for this issue to add a WeightedLeastSquaresTest.py which try to expose the general level by which we would like to support weighted least squares fitting in Mantid. This systemtest will complement UnweightedLeastSquaresTest.py. These two systemtests could also be a place where we could add fitting examples which we would like to support but correctly don't.

PR #14278 (2015-11-17): Add weighted ls fitting system tests.
Fixes #13348

### To test
* check that systemtest pass
* check that text in systemtest reads well and is clear
* check 1st good example added to this test
closed 2015-11-17 by FedeMPouzols