mantidproject/mantid #13355: Test the mediawiki doctests nightly

NickDraper, 2015-08-11:
issue #13310 an pull request #13311 added doctests for mediawiki pages

This is to add them to the ?nightly automated build

It involves running the script 
``` 
python.exe TestWikiPython.py -o C:\Mantid\Code\Mantid\docs\source\mwTests
```

and then running the doctests job as normal.the path in C:\Mantid\Code\Mantid\docs\source\mwTests must be within the source directories of the sphinx help, but does not need to exist in advance.

Feel free to move the TestWikiPython.py script to another location if it makes more sense for it to be there, it is a standalone script, and should be ignorant of it's location.

The build servers will need there HTTP_Proxy environment setting set if necessary to get a network connection out to the mediawiki site.

NickDraper, 2015-11-04:
@FedeMPouzols This might be a good intro task to get to know more of the build infrastructure.

FedeMPouzols, 2015-11-16:
The plan is to add this command at the beginning of the *nightly* doctest builds so the special doctests added by `TestWikiPython.py` run together with the standard doctests.

NickDraper, 2015-11-16:
Sounds good

PR #15192 (2016-02-22): Test the doc tests from mediawiki examples nightly
Fixes #13355.

**To test**:
- Check the Jenkins job: http://builds.mantidproject.org/view/All/job/master_doctests_wikitests. The build has been working well for a few days.
- Initially we had one failure in one of the SliceViewer examples (outdated expected output). Currently all tests pass.
- Check that it is actually testing the special mediawiki tests (the ones `mwTests/mwTest_...`). There should be >=1004 tests in total.
- I think the current location of TestWikiPython.py is fine. An alternative could be to move it to `docs/` where you als [... cut]
closed 2016-02-22 by martyngigg