mantidproject/mantid #13372: Minimum width of algorithm browser widget is larger than it should be

DanNixon, 2015-08-13:
The minimum width is taken from that of the contents of the drop down menu which i snow much larger that the aliases are also in there.

DimitarTasev, 2016-07-19:
@DanNixon @NickDraper 
I have tried to replicate the issue and this is the result
[image]

In the code [that initialises the Algorithm Browser Widget](https://github.com/mantidproject/mantid/blob/master/MantidQt/MantidWidgets/src/AlgorithmSelectorWidget.cpp#L35) there are no policies/settings for the width, and I think the algorithms browser wouldn't be very useful if it could get any smaller.

If you meant a different part/problem could you please direct me to it?

DanNixon, 2016-07-19:
@DimitarTasev I'll try a new build and check, it's an old issue so it may not be applicable anymore (there may be a difference between platforms too as I only use Linux).

closed 2017-01-16 by NickDraper