mantidproject/mantid #13383: Add QENS peaks to CreateSampleWorkspace

DanNixon, 2015-08-13:
Add two default peak types, one for standard QENS, one for QENS with tunnelling side peaks.

stuartcampbell, 2015-08-13:
Just wondering, do you have any more details on what a 'standard QENS' function actually is ?

DanNixon, 2015-08-13:
The function I would have used for this is:
```
name=Lorentzian,FWHM=0.3,PeakCentre=$PC5$,Amplitude=0.8;name=Lorentzian,FWHM=0.1,PeakCentre=$PC5$,Amplitude=1;name=LinearBackground,A0=0.1
```

I agree defining a "standard" may be difficult, the main reason for the issue is to add the tunnelling example, the other I just want to add so I don't have to have the function defined in multiple tests that I want to replace the file loading in.

PR #13385 (2015-08-19): Add QENS peaks to CreateSampleWorkspace
Fixes #13383.

To test:
- Run `CreateSampleWorkspace` with:
 - `Function` set to either `Quasielastic` or `Quasielastic Tunneling`, 
 - `XMin`=-0.5
 - `XMax`=0.5
 - `BinWidth`=0.01
- See that the functions are reasonable

Release notes updates [here](http://www.mantidproject.org/index.php?title=ReleaseNotes_3_5_Framework_Changes&diff=24977&oldid=24975).
closed 2015-08-19 by FedeMPouzols