mantidproject/mantid #13390: Q1DWeighted normalization / weighting

SimonHeybrock, 2015-08-14:
The `Q1DWeighted` algorithm uses a peculiar 2-stage weighting / normalization scheme: In a first step averages of all pixels in the same circle are computed. In a second step all circle-averages are averaged, giving each circle the same weight.

This is quite obviously not correct. The EQSANS instrument scientist wants to stay with this scheme however. Instrument scientists of other SANS instruments that use the algorithm have been contacted, but I got no reply so far.

 - The algorithm should be renamed to something like `EQSANSQ1DWeighted`, and the documentation must be updated to warn potential users of the non-standard weighting.
 - A `Q1DWeighted` algorithm with a standard weighting should be created.

I have postponed these changes until I hear from other instruments.

NickDraper, 2017-06-29:
This is only used by EQSANS, and is not used at ISIS.

I've removed the Easy label as this involves a potential breaking change (renaming an algorithm) so would not be suitable for a new starter.

gvardany, 2019-03-07:
This is no longer relevant after PR #24699. It was understood that this weighting scheme is also correct provided that the resolutions are also averaged in the similar way. Alternatively one can use Q1D for one step averaging.

closed 2019-03-07 by gvardany