mantidproject/mantid #13397: MDBoxBase incomplete copy constructor.

SimonHeybrock, 2015-08-17:
The copy constructor of `MDBoxBase` sets new extents, but does not call `calcVolume()`. I am pretty sure it should (and even the Doxygen comment claims that it should).

Add unit test that exposes bug, fix the bug.

davidfair, 2016-07-12:
@SimonHeybrock I've looked into this issue, currently it copies the value computed by `calcVolume()` into `m_inverseVolume` which is copied into the new object. 
Is it worth recalculating this value on the new object or just copying the value as the array it is calculated from is also copied?

SimonHeybrock, 2016-07-13:
@DavidFair You are right, I must have overlooked that, thanks for pointing it out. I am closing this then.

PR #16866 (2016-07-14): Update MDBoxBase copy constructor docs to reflect behaviour
Description of work.
Updated documentation on MDBoxBase copy constructor to correctly reflect its behaviour.
This is a minor documentation only change so release notes were not updated. 

Fixes #13397 .

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with sm [... cut]
closed 2016-07-13 by SimonHeybrock