mantidproject/mantid #13408: ISIS powder diffraction workflow

FedeMPouzols, 2015-08-17:
Understand what is the workflow in the ISIS powder diffractions scripts.

This may include a diagram in `.dot` format, on paper, etc. that would at some point become part of the documentation of the ISIS powder diffraction algorithm(s).

As part of this:
- Understand what Mantid algorithms are currently being used
- What manual or low level is being done that could be replaced with existing Mantid algorithms.
- What functional blocks could become modified or new Mantid algorithms.

shahrozahmed1, 2015-09-23:
Fixes #13408

I have uploaded the [general workflow](https://github.com/mantidproject/documents/blob/dfafba29d6a2d72012de3dce8d2b3e46ac531a44/Design/PowderDiffraction_ISIS/PowderDiff_General_Workflow_png.png) and [algorithms-used workflow](https://github.com/mantidproject/documents/blob/dfafba29d6a2d72012de3dce8d2b3e46ac531a44/Design/PowderDiffraction_ISIS/PowderDiff_AlgoritmUsed_FlowChart_png.png) in  mantidproject/documents/Design/PowderDiffraction_ISIS directory.

@FedeMPouzols

FedeMPouzols, 2015-09-25:
This complements the PowDiff document very nicely.

closed 2015-09-25 by FedeMPouzols