mantidproject/mantid #13409: ISIS powder diffraction: instruments other than HRPD

FedeMPouzols, 2015-08-17:
The examples provided with the scripts use HRPD files. That should be enough for a start (for example for #13405 and #13407).

Will they work also with GEM and POLARIS, without errors/warnings?

If the answer is yes, then there should be additional test cases, for example as part of #13407, or extending it.

SpencerHowells, 2015-08-17:
The OSIRIS high angle detector bank also does powder diffraction

stuartcampbell, 2015-08-17:
And an expertly designed detector bank it is! :smile:

FedeMPouzols, 2015-08-17:
@SpencerHowells , @stuartcampbell : thanks for your feedback. Are the "Aziz scripts" used for that expertly designed bank of OSIRIS?

stuartcampbell, 2015-08-19:
@FedeMPouzols @SpencerHowells I thought that there was some special indirect diffraction scripts - or at least that was how it used to be (Sp will know better).  Here we are going to try using the standard powder diffraction routines for our diffraction banks on indirect instruments.  Also don't forget that there are diffraction banks on IRIS as well - They don't get dedicated experiments, but probably should be processed the same way.

DanNixon, 2015-08-19:
@stuartcampbell OSIRIS has it's own diffraction reduction that does vanadium corrections similar to other diffraction reduction workflows (AFAIK).

There is a general purpose indirect diffraction script but it is rarely used on OSIRIS. IRIS and TOSCA (and VESUVIO) just use the general purpose one.

@FedeMPouzols Maybe it would be worth talking to Sanghamitra about OSIRIS, she had a few issues with the current reduction workflow that she mentioned a while ago.

FedeMPouzols, 2015-08-26:
So  @SpencerHowells and @DanNixon would like to have OSIRIS support. @DanNixon says "there is a difference in that to have the large d-range of other diffractometers you need to change wavelength on OSIRIS, hence multiple runs must be used for a single experiment". When we get to this, check the details with Franz Demmel.

SpencerHowells, 2015-09-09:
VESUVIO will also want to do diffraction analysis.

FedeMPouzols, 2016-05-27:
Scripts for PEARL are now included for v3.7: http://docs.mantidproject.org/nightly/api/python/techniques/PearlPowderDiffractionISIS-v1.html

DavidFair, 2016-09-12:
@FedeMPouzols Is this issue still applicable or can it be closed?

DavidFair, 2017-01-27:
This issue can be closed as the migration of various instruments is being tracked in #17831

closed 2017-01-27 by DavidFair