mantidproject/mantid #13417: ScriptWindow + Console MantidPlot launch

OwenArnold, 2015-08-18:
We have a minor usability issue with the new changes. This occurs when you launch MantidPlot with an argument for the path of a script to open.

* Run MantidPlot like so ```MantidPlot.app/Contents/MacOS/MantidPlot {path_to_script}```
* MantidPlot opens with the ScriptWindow populated with the script set in {path_to_script}
* Now close MantidPlot
* Now run MantidPlot using the same first command again
* Now the script window has opened with two tabs pointing to the same script. 

I think there should be unique check on the paths for each script in the script window upon open. I think what's happening is that the new mechanism is living independently from the old console launch options.

ElliotAOram, 2015-08-18:
@OwenArnold Do you know if this is MacOS specific, I can't seem to reproduce on Windows.
Regardless I agree that a check for unique file paths is a good idea!

OwenArnold, 2015-08-18:
You're right, this doesn't happen on Windows

PR #13603 (2015-09-14): Script window conflicting with console launch
Fixes #13417

Scripting window now checks to see if the command line argument (if it is a script) matches a previously used argument to avoid opening duplicate scripts with calls from the command line.

# To Test
* Run Mantid from the command line with the name of the script as an argument 
* Ensure the script opens automatically in a scripting window
* Close Mantid
* Run Mantid again from the command line with the same script as an argument
* Ensure that this does not open a new copy of the script and only one copy is open in the scripting window
closed 2015-09-14 by picatess