mantidproject/mantid #13456: LoadEventNexus is duplicating events

SimonHeybrock, 2015-08-20:
To reproduce:

- Load -> select a Nexus file (tested with `LET00006278.nxs`).
- pick a spectrum range, for example `SpectrumMin=10` and `SpectrumMax=20`.
- do the same again, to a different output workspace, but now with deselected `Precount`.

The workspace from the first load shows *twice* the number of events --- the event number in each spectrum is *doubled*. This does not happen when the full range is loaded.

This was discovered together with @MichaelWedel, tested on Mac and Ubuntu.

stuartcampbell, 2015-08-20:
Do you get the same issue with SNS event files ?

SimonHeybrock, 2015-08-20:
We tried with TOPAZ and PG3, when we select a range we just get 0s, *no* events seem to be loaded.

SimonHeybrock, 2015-08-25:
Update: This issue does not seem to affect SNS event files. The comment above regarding no events being loaded is due to #13475.

NickDraper, 2015-11-19:
cut down test
``` python
ws1 = Load(Filename='OFFSPEC00036416.nxs', SpectrumMin=10, SpectrumMax=20, Precount = True)
ws2 = Load(Filename='OFFSPEC00036416.nxs', SpectrumMin=10, SpectrumMax=20, Precount = False)
if ws1.getNumberEvents() == ws2.getNumberEvents():
    print "passed - all is good"
else:
    print "failed"
    print "ws1 events:", ws1.getNumberEvents()
    print "ws2 events:", ws2.getNumberEvents()
```

NickDraper, 2015-11-20:
Looking into it this will actually affect ANY event nexus file where the number of cores is more than twice the number of banks. So all ISIS nexus files, and any SNS ones with small numbers of banks.

Of course it only occurs when you have selected a smaller range of spectra, either by max and min or the list.

@peterfpeterson @stuartcampbell  Flagging you so you are aware of this one!  Is this likely to hit any SNS instruments?

closed 2015-11-27 by LamarMoore