mantidproject/mantid #13475: LoadEventNexus interprets spectrum index as detector ID

SimonHeybrock, 2015-08-25:
`LoadEventNexus` has the optional `SpectrumMin`, `SpectrumMax`, and `SpectrumList` properties. For ISIS files this is really the spectrum (but there are other issues, see #13456 and #13457). For non-ISIS files, however, this is interpreted as the *detector ID*. This may be the desired behavior, but the documentation and GUI should be fixed.

NickDraper, 2016-02-19:
For the SNS they do not have a concept of a spectrum ID (as they never group detectors in hardware), therefore the detector id does map to spectrum ID.

NickDraper, 2016-02-19:
rst page updated (and mistakenly submitted directly to master using a key combination I was not aware of)

https://github.com/mantidproject/mantid/blob/master/docs/source/algorithms/LoadEventNexus-v1.rst

SimonHeybrock, 2016-02-22:
@NickDraper Did you also update the GUI (field names and tool tips)?

SimonHeybrock, 2016-02-29:
I will reopen this until it is fixed. This is also part of the 'spectrum number' cleanup.

NickDraper, 2016-04-04:
Spectrum index is a valid concept in nexus files, and does map to those two concepts in this case.

closed 2016-04-04 by NickDraper