mantidproject/mantid #13488: Engg: figure out if/how to write focused data into OpenGenie 'his' files

FedeMPouzols, 2015-08-26:
Nobody seems to know or remember this `his` format which I understand is just a binary dump of histogram data (with maybe? 2/3 data columns), similar to what is produced by `SaveGSS` and `SaveFocusedXYE`.

There would be alternatives or workarounds but if this turns out to be doable it will be convenient for interfacing with legacy OpenGenie routines.

FedeMPouzols, 2015-08-26:
Ah ok, this clarifies a bit how they write these `his` files: https://github.com/stuartcampbell/opengenie-scripts/blob/master/ENGIN/FOCUS25.COM

FedeMPouzols, 2015-08-26:
And there's definitely a solution for this. Saurabh just sent me an email: there's an ascii format of the 'his' files that the ENGIN-X OpenGenie scripts can load. That should be enough. It could become an algorithm `SaveFocusedOpenGenieAscii`.

FedeMPouzols, 2015-08-27:
Renaming this from `Engg: figure out if/how to write focused data into OpenGenie 'his' files` to `Engg GUI: add support for saving OpenGenie ascii his files`.

FedeMPouzols, 2015-09-09:
Reopening as this seems to have been closed from #13452 which is unrelated.

shahrozahmed1, 2015-10-15:
This has been fixed in another issue/PR.

PR #13452 (2015-09-09): Correct the input parameters for the fit
Fixes #13488
Fixes #13402

The changes have a scientific validation by Richard Heenan, so we just want to make sure that execution runs without errros.

# For testing

Please find the relevant user files, etc here: \\olympic\Babylon5\Scratch\Anton\Testing\13448_fix_scale_and_shift_for_front_detector

The folder also contains an installer

The user file is: USER_SANS2D_151U_2p4_4m_M3_Banuelos_8mm_ECcell_changer.txt
The data files are:
Sample Scattering    SANS2D00031096.nxs
Sample Transmission   SANS2D00031097.nxs
Sample Direct   SANS2D00031092.nxs

Can  [... cut]
closed 2015-10-16 by NickDraper