mantidproject/mantid #13498: Mock-up for the fully features Engg GUI focus tab

FedeMPouzols, 2015-08-27:
Prepare a mock-up for the extended focus tab, to narrow down all the details from yesterday's discussion and notes.

We usually use the balsamiq tool.

shahrozahmed1, 2015-09-07:
@FedeMPouzols  The mock-up for the `Engg GUI: focus tab` can be found  [here](https://github.com/mantidproject/documents/blob/1d2637c2e384de996396beeddefb5549ba967cab/Design/ENGIN-X/GUI/Mockup_Engineering_Diffraction_FocusTab.pdf):
`documents/Design/ENGIN-X/GUI/Mockup_Engineering_Diffraction_FocusTab.pdf`

FedeMPouzols, 2015-09-14:
The tab looks good and more complete now. I think it needs a 'focus' push button inside the group 'Focus Cropped' and also the group 'Focus Texture', or I might be forgetting something from the original plans.

shahrozahmed1, 2015-09-18:
well I personally think that it looks clean with just the one focused button on the bottom but I have created both Mock-up and you can choose one from the following or maybe ask Saurabh. 
 Mock-Up: [Option 1](https://github.com/mantidproject/documents/blob/a710de6d9d5207b9f05c7c18eeea64287311c6bf/Design/ENGIN-X/GUI/Engineering_Diffraction_Analysis%20option_1.pdf) - [Option 2](https://github.com/mantidproject/documents/blob/a710de6d9d5207b9f05c7c18eeea64287311c6bf/Design/ENGIN-X/GUI/Engineering_Diffraction_Analysis%20option_2.pdf)

FedeMPouzols, 2015-09-25:
Looks good. Let's see what variant(s) are used in the end.
You can also upload the xml files from balsamiq to the same place on github so you keep copies of the sources in case you want to edit these mock-ups at a later stage.

closed 2015-09-25 by FedeMPouzols