mantidproject/mantid #13508: Need a CompactMD algorithm

OwenArnold, 2015-08-27:
This should be relatively easy to do. All that the compact command does is find the edges where we have the first non-zero data and then will run CutMD to trim the workspace to that size and output it as a new workspace. Ask Owen to explain this further if needed. See the horace documentation for Compact [here](http://horace.isis.rl.ac.uk/Reshaping_etc#compact). This algorithms should work like that.

Caveats:

* Must only work on regular binned input workspaces. Start by assuming the inputs must be IMDHistoWorkspaces
* Should try to maintain the existing bin widths in each dimension as not to distort the data too much.

MattKing06, 2015-10-01:
This issue is being blocked by #13796 as it uses IntegrateMDHistoWorkspace. Once #13796 has been merged, this algorithm can be developed and tested.

closed 2015-10-15 by picatess

OwenArnold, 2015-10-15:
@matthew-d-jones You should take a look at this completed algorithm as it falls under the Horace/Mantid/VATES commands.

matthew-d-jones, 2015-10-15:
@OwenArnold Ok, thanks.

PR #13710 (2015-10-15): Create CompactMD Algorithm
Fixes #13508

Create a CompactMD algorithm based on [Horace](http://horace.isis.rl.ac.uk/Reshaping_etc#compact) compact command.