mantidproject/mantid #13551: Figure out what's best to do with system test HRPDPowderDiffraction

FedeMPouzols, 2015-09-04:
It was found  (see #13531) that we have [this system test](https://github.com/mantidproject/systemtests/blob/master/SystemTests/AnalysisTests/HRPDPowderDiffraction.py) from over 4 years ago. It may need:
 - update
 - be moved and/or merged with other powder diffraction tests, maybe as a particular test case if it is
   useful
 - it may not make sense as a useful reduction any longer, but still I wouldn't remove it unless there's
   any issues along the road. In principle the test is doing sensible operations and testing several
   of the algorithms that are important for the "Aziz scripts".

This test looks to me as a simplified or particular case reduction but similar
to what the "aziz" scripts do. So it can be useful for the other issues related 
to current ISIS powder diffraction update effort (for example #13408).

This issue is to try to clarify the role and potential usefulness of this old system test when 
we meet with instrument scientists. Keep this in the list of things to discuss.

Tip: some of the checks that are done in this test can be good inspiration for #13538.

shahrozahmed1, 2015-10-09:
@FedeMPouzols Now that #13730 has been merged, should this be closed?

FedeMPouzols, 2015-10-09:
I'd keep it in the list of points to discuss. This script is much simpler to interpret and can be helpful to understand the workflow of the "Aziz" scripts. A note in the general PowderISIS readme would be handy as a reminder.

closed 2015-10-09 by FedeMPouzols