mantidproject/mantid #13572: Improvements to SavePlot1DAsJson

peterfpeterson, 2015-09-08:
There are some small changes:

1. For the units, use the result of `axis.symbol().latex()` or fall through to `axis.symbol()` if it generates an exception. This should give proper mark-up for Angstroms.
1. If the units are empty, don't include the parenthesis (e.g. `Spectrum ()` should be `Spectrum`)
1. The `PlotName` property doesn't belong. Please remove it.
1. The title for each spectrum should be the workspace title and the spectrum number. `"%s - spectrum %d" % (workspace.getTitle(), workspace.getSpectrum(i).getSpectrumNo())`
1. The labels for the various spectra should be unique. A useful setting would be `"%s_spectrum_%d" % (workspace.getName(), workspace.getSpectrum(i).getSpectrumNo())` (negotiable).
1. There needs to be an additional argument, `MainOutputSpectrum`, to select a spectrum to label as `main_output`. This is required by the live monitor so it can select which spectrum to plot. This should default to the first spectrum if it is not set by the user. `Property.EMPTY_INT` and `self.isEmpty` will probably be helpful in implementing this.

This is a follow on to #10234.

peterfpeterson, 2015-09-08:
@mdoucet should look over this list.

yxqd, 2015-09-17:
Proposed json data format for the plot:

plot = {
  data: {
label1: (N, 3) array,
label2: (N, 3) array,
…},
  datalabels: [label2, label1, …],
  axes: {
xlabel = …,
xunit = …,
ylabel = …,
yunit = …,
…}
}

yxqd, 2015-10-02:
Mathieu requested a change of the data object. Instead of 

```plot = {
  name: <name>,
  type: "histogram" or "point",
  data: {
label1: (N, 3) array,
label2: (N, 3) array,
…},
  axes: {
xlabel = …,
xunit = …,
ylabel = …,
yunit = …,
…}
}```

he prefers

```plot = {
 <name>: {
  type: "histogram" or "point",
  data: {
label1: (N, 3) array,
label2: (N, 3) array,
…},
  axes: {
xlabel = …,
xunit = …,
ylabel = …,
yunit = …,
…}
}}```

PR #13755 (2015-10-13): Improve saveplot1dasjson
This fixes #13572
closed 2015-10-13 by mdoucet