mantidproject/mantid #13575: TOPAZ slice viewer issue

peterfpeterson, 2015-09-08:
To reproduce:

1. `Load("T295K_combined.nxs", OutputWorkspace="T295K_combined")`
2. `Show Slice Viewer` of this dataset
3. Change the color scale from `logarithmic` to `linear`
4. Get a `std::bad_alloc` or a proper segfault

rosswhitfield, 2015-09-08:
Example script on how this is generated. The `dataMD` and `normMD` work just fine but the `normalized` after DivideMD is the one with the problem plotting. Taken from `/SNS/TOPAZ/IPTS-14787/shared/visualization/processTOPAZ_IPTS-14787.py`. If you run this I suggest reducing the `runs` to just one.
``` python
import sys
sys.path.append("/opt/mantidnightly/bin")
from mantid.simpleapi import *
import mantid

instrument_name ='TOPAZ'
datadir = r'/SNS/TOPAZ/IPTS-14787/data/'

#Vanadium and calibration files
geometryfile=r'/SNS/TOPAZ/IPTS-14787/shared/calibration/TOPAZ_2015B.DetCal'
groupingfile=r'/SNS/TOPAZ/IPTS-14787/shared/calibration/grouping-no58.xml'
calibrationdir =r'/SNS/TOPAZ/IPTS-14787/shared/calibration/'
vanadium = calibrationdir + 'TOPAZ_12387_event.nxs'
nosample = calibrationdir + 'TOPAZ_12389_event.nxs'

#Sample temperature in K 
T = 100
#UB matrix
UBfile=r'/SNS/TOPAZ/IPTS-14787/shared/100K/TOPAZ_12413_R.mat'    # need to add integration directory
#run numbers
runs=range(12413,12423)
#runs.append(11896)
#Output directory
wd=r'/SNS/TOPAZ/IPTS-14787/shared/visualization/combined/'   # need to add integration directory
#wd=r'/SNS/TOPAZ/IPTS-14803/shared/visualization/temp/'   # need to add integration directory
outfile_name ='T'+str(T) + 'K'

BKG=Load(Filename=nosample)
LoadIsawDetCal(InputWorkspace=BKG, Filename=geometryfile)
BKGPC=BKG.getRun().getProtonCharge()
#vanadium processing
try:
    sa=Load(calibrationdir+'solidAngle12387.nxs')
    flux=Load(calibrationdir+'spectra12387.nxs')
except:    
    rawVan=Load(Filename=vanadium)
    LoadIsawDetCal(InputWorkspace=rawVan, Filename=geometryfile)
    #BKG=Load(Filename=nosample)
    #LoadIsawDetCal(InputWorkspace=BKG, Filename=geometryfile)
    #BKG=NormaliseByCurrent(BKG)
    rawVanPC=rawVan.getRun().getProtonCharge()
    temp=BKG*(rawVanPC/BKGPC)
    rawVan=Minus(LHSWorkspace=rawVan, RHSWorkspace=temp)
    #Absorption correction added on November 03, 2014
    rawVan=CropWorkspace(InputWorkspace=rawVan,XMin='500.0',XMax='16600.0')
    rawVan=AnvredCorrection(InputWorkspace=rawVan,
      LinearScatteringCoef='0.367',LinearAbsorptionCoef='0.366',
      Radius='0.20',OnlySphericalAbsorption='1',PowerLambda ='0')
    #scale = 1.0
    #rawVan=rawVan*scale
    rawVan=ConvertUnits(InputWorkspace=rawVan,Target='Momentum')
    MaskBTP(Workspace=rawVan,Bank="10-12,15,20-21,24-25,30-32,35,40-45,50-57,58,59")
    MaskBTP(Workspace=rawVan,Pixel="0-15,240-255")
    MaskBTP(Workspace=rawVan,Tube="0-15,240-255")
    #Change the Xmin and Xmax in units of Momentum 2pi/lambda
    rawVan=CropWorkspace(InputWorkspace=rawVan,XMin='1.50',XMax='32.0')
    #The sa is a spatial sensitivity correction
    sa=Rebin(InputWorkspace=rawVan,Params='1.50,32,32.0',PreserveEvents='0')
    SaveNexus(InputWorkspace=sa, Filename=calibrationdir+"solidAngle12387.nxs")
    #Check grouping file. Need to match the current instrument config.
    #The flux has the event data corrected for absorption
    flux=Rebin(InputWorkspace=rawVan,Params='1.50,32,32.0',PreserveEvents='1')
    flux=GroupDetectors(InputWorkspace=flux,MapFile=groupingfile)
    #DeleteWorkspace(rawVan)
    #DeleteWorkspace(BKG)
    AnalysisDataService.remove( rawVan.getName() )
    #AnalysisDataService.remove( BKG.getName() )
    flux=CompressEvents(InputWorkspace=flux,Tolerance=1e-4)
[... 45 lines cut ...]
                       QConversionScales="HKL",
                       MinValues="-5.01,-5.01,-15.025",
                       Maxvalues="5.01,5.01,15.025")
        filename=wd+instrument_name+'_'+str(r)+'_MDE.nxs'
        SaveMD(InputWorkspace=md,Filename=filename)
        #DeleteWorkspace(data)
        #AnalysisDataService.remove( data.getName() )

    a,b=MDNormSCD(InputWorkspace=md,FluxWorkspace=flux,SolidAngleWorkspace=sa,
                  AlignedDim0="[H,0,0],-5.01,5.01,501",
	              AlignedDim1="[0,K,0],-5.01,5.01,501",
	              AlignedDim2="[0,0,L],-15.025,15.025,601")
	
    #DeleteWorkspace(md)
    #AnalysisDataService.remove( md.getName() )
    
    if r==runs[0]:
        dataMD=CloneMDWorkspace(a)
        normMD=CloneMDWorkspace(b)
    else:
        dataMD+=a
        normMD+=b

normalized=DivideMD(dataMD,normMD)
```

peterfpeterson, 2015-09-08:
I think this is just pointing out how the invalid data is created.

peterfpeterson, 2015-09-08:
@OwenArnold Have you seen something like this before?

OwenArnold, 2015-09-09:
@peterfpeterson  I thinks someone recently added something to the color scale for power scales in the SliceViewer. It might have been @matthew-d-jones ?

PR #13589 (2015-09-10): Fix Slice Viewer Exception for normalized data
Fixes #13575 There was a -inf in data

    Load("/SNS/TOPAZ/IPTS-14803/shared/visualization/combined/T295K_combined", OutputWorkspace="T295K_combined")
    Show Slice Viewer of this dataset
    Change the color scale from logarithmic to linear
    Don't get a std::bad_alloc or a proper segfault
closed 2015-09-10 by rosswhitfield