mantidproject/mantid #13596: Log filtering for status and period does not appear to work by default

NickDraper, 2015-09-10:
We introduced log filtering as on by default in R2.2, but it now longer appears to be happenening.

```python
ws = mtd["POLREF00013681"]
r = ws.run()
fs = r.getLogData("MethanTemp")
print type(fs)
print fs.size()

print fs.valueAsString()
```
returns the full unfiltered log (the 38 value should not be present, and the length should be shorter)
```
<class 'mantid.kernel._kernel.FloatTimeSeriesProperty'>
15
2015-Jun-01 09:16:39  25
2015-Jun-01 09:22:40  25
2015-Jun-01 09:22:41  25
2015-Jun-01 09:56:58  25
2015-Jun-01 09:56:58  25
2015-Jun-01 15:15:59  33
2015-Jun-01 15:22:45  38
2015-Jun-01 15:30:25  33
2015-Jun-01 15:39:35  34
2015-Jun-01 15:39:35  34
2015-Jun-01 15:39:55  34
2015-Jun-01 15:39:55  34
2015-Jun-01 15:41:02  34
2015-Jun-01 15:41:02  34
2015-Jun-01 15:58:36  34
```

Investigate why and when this was introduced. (it is possible it is important for filtering or the SNS).

NickDraper, 2015-09-10:
@abuts Is interested in this as well

martyngigg, 2015-09-16:
The bug here seems to be with the `ISISRunLogs` class, specifically the [addPeridoLog](https://github.com/mantidproject/mantid/blob/master/Code/Mantid/Framework/DataHandling/src/ISISRunLogs.cpp#L54) method.

I think its implementation should be:
```c++
 auto periodLog = m_logParser->createPeriodLog(period);
  LogFilter *logFilter(NULL);
  const TimeSeriesProperty<bool> *maskProp(NULL);
  try {
    auto runningLog =
        exptRun.getTimeSeriesProperty<bool>(LogParser::statusLogName());
    logFilter = new LogFilter(*runningLog);
  } catch (std::exception &) {
    g_log.warning(
        "Cannot find status log. Logs will be not be filtered by run status");
  }

  // If there is more than 1 period filter the logs by period as well
  if (m_logParser->nPeriods() > 1) {
    if (logFilter) {
      logFilter->addFilter(*periodLog);
      maskProp = logFilter->filter();
    } else
      maskProp = periodLog;
  } else {
    maskProp = logFilter->filter()
  }
  // Filter logs if we have anything to filter on
  if (maskProp)
    exptRun.filterByLog(*maskProp);
  delete logFilter;

  exptRun.addProperty(periodLog);
  exptRun.addProperty(m_logParser->createCurrentPeriodLog(period));
  try {
    exptRun.addLogData(m_logParser->createAllPeriodsLog());
  } catch (std::runtime_error &) {
    // Already has one
  }
```

The sample logs GUI also has contradictory behaviour to the framework. In [MantidUI](https://github.com/mantidproject/mantid/blob/master/Code/Mantid/MantidPlot/src/Mantid/MantidUI.cpp#L2579) the filter is artificially altered to add an entry when the first time of the filter is after the first time of the log. The framework **does not** do this.

Finally, the `TimeSeriesProperty` class has a highly inconsistent interface. When filtered using the `filter*` methods some members respect the filter and return answers based on the filtered set and others do not.

martyngigg, 2017-03-09:
[image] [Fix Period Log Filtering in Python](https://trello.com/c/czKkcGXq/321-fix-period-log-filtering-in-python)

martyngigg, 2019-07-16:
This was fixed in #25911

closed 2019-07-16 by martyngigg