mantidproject/mantid #13601: Create default normalization dropdown for VSI

picatess, 2015-09-10:
We want a normalization dropdown for the VSI. There the user can specify a default normalization

OwenArnold, 2015-10-06:
* Layout doesn't seem right. Too much vertical padding.
[image]
* Items should be in group box.
* No need for word initial. I think this is misleading.
* VSI help [page](http://www.mantidproject.org/VatesSimpleInterface_v2) would need updating for all these changes.
* Auto == AutoSelect. We should dynamically put some text next to what Auto means (ask Owen for details)
* Normalization property doesn't seem to update when I change it on the new top level menu.
* Since the VSI and the SliceViewer need the same control for this, lets create one that can really be reused by both.

AntonPiccardoSelg, 2017-10-05:
Some work had been done for the VSI, but we thought that this could/should be shared with the SliceViewer

AntonPiccardoSelg, 2017-10-05:
The normalization can be specified in the pipeline browser. It is very unlikely that resources will be allocated towards this task in the future. Hence I am closing this issue

closed 2017-10-05 by AntonPiccardoSelg